whitecatboard/Lua-RTOS-ESP32 #1: mtx_lock() occurs in two times

edbek, 2016-11-16:
mtx_lock() function occurs in two times in pthread_create(). So it should be?

```
mtx_lock(&thread->init_mtx);

    // Create related task
    res = xTaskCreate(
            pthreadTask, "lthread", stacksize, taskArgs,
            tskDEF_PRIORITY, &xCreatedTask
    );

    if(res != pdPASS) {
        // Remove from thread list
        list_remove(&thread_list,*id);
        free(taskArgs);
        free(thread);

        if (res == errCOULD_NOT_ALLOCATE_REQUIRED_MEMORY) {
            errno = ENOMEM;
            return ENOMEM;
        } else {
            errno = EAGAIN;
            return EAGAIN;
        }
    }

    mtx_lock(&thread->init_mtx);
```

jolivepetrus, 2016-11-16:
This is correct, it's mandatory for Lua RTOS pthread management.

After the creation of the new task related to new thread we need to wait for the initialization of critical information that is provided by new thread:

* Allocate Lua RTOS specific TCB parts, using local storage pointer assigned to pthreads
* The thread id, stored in Lua RTOS specific TCB parts
* The Lua state, stored in Lua RTOS specific TCB parts

This is done by pthreadTask function, who releases the lock when this information is set.

We have document this in our last commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/fb44da2fe2c93128eb833b042ccf13838a5ba4eb

jolivepetrus, 2016-11-23:
Let me to close this issue. No feedback were provided.

closed 2016-11-23 by jolivepetrus