whitecatboard/Lua-RTOS-ESP32 #2: case sensivity issue including "FreeRTOS.h"

RonnyA, 2016-12-04:
Several files has wrong case on the FreeRTOS.h file so that the code doesnt compile on Linux without modifying the source files.

This includes:
        components/lua_rtos/drivers/lmic_hal.c
        components/lua_rtos/luartos.h
        components/lua_rtos/pthread/attr.c
        components/lua_rtos/unix/getcwd.c

jolivepetrus, 2016-12-05:
Thanks for reporting this. This have been solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/faaae388f333e1e779c754b65bbcc7459cc04439.

jolivepetrus, 2016-12-12:
Thanks for reporting this. This have been solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/edcfb571a952cb46a75f884d982d4fb0935948bd.

closed 2016-12-13 by jolivepetrus