whitecatboard/Lua-RTOS-ESP32 #3: ADC channel sanity check not working?

MrZANE42, 2016-12-12:
First of: Excuse me if I'm wrong I haven't tried the code I just happened to see something strang while looking through the code in the repo.
In Lua-RTOS-ESP32/components/lua_rtos/drivers/adc.c it's written (in a couple of places):
if (!((1 << channel) && CPU_ADC_ALL))

I'm quite sure that that should be written as so:
if (!((1 << channel) & CPU_ADC_ALL))

That is: & instead of &&

jolivepetrus, 2016-12-12:
https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/edcfb571a952cb46a75f884d982d4fb0935948bd

jolivepetrus, 2016-12-12:
Thanks for reporting this. This have been solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/edcfb571a952cb46a75f884d982d4fb0935948bd.

closed 2016-12-13 by jolivepetrus