whitecatboard/Lua-RTOS-ESP32 #5: adc.c compile error

loboris, 2016-12-14:
When compiling on Linux I've got the following error:
```
In file included from /home/LoBo2_Razno/ESP32/Lua-RTOS-ESP32/components/lua_rtos/Lua/modules/adc.c:39:0:
/home/LoBo2_Razno/ESP32/Lua-RTOS-ESP32/components/lua_rtos/Lua/modules/adc.h:34:25: fatal error: drivers/CPU.h: No such file or directory
 #include "drivers/CPU.h"
                         ^
compilation terminated.
/home/LoBo2_Razno/ESP32/esp-idf/make/component_wrapper.mk:176: recipe for target 'Lua/modules/adc.o' failed
```
After changing (in _components/lua_rtos/Lua/modules/adc.h/adc.h_)
```
#include "drivers/adc.h"
#include "drivers/CPU.h"
```
to
```
#include <drivers/adc.h>
#include <drivers/cpu.h>
```
it builds without errors.

By the way, great work, I'm testing the wifi branch, for now everything works...

jolivepetrus, 2016-12-14:
Thanks for report this. This is solved in commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/2ff5a97e3f0a767d63d04af9acb92d77ee59b3ae.

jolivepetrus, 2016-12-14:
Please, take note that actually is not clear what's the ADC internal voltage reference. It seems that is near to 1.1 Volts. We are waiting information about this from Espressif.

closed 2016-12-14 by jolivepetrus