whitecatboard/Lua-RTOS-ESP32 #8: Error when calling pthread_cond_signal() function

edbek, 2016-12-20:
Inside pthread_cond_signal() is called each time mtx_init(). 
mtx_init() is created the semaphore in any case:

`mutex->sem = xSemaphoreCreateBinary ();
`

This condition should be avoided:

`if ( !mutex-> sem ) mutex->sem = xSemaphoreCreateBinary ();`

In addition, all of the objects by malloc () must be reset. Used calloc () or memcpy(0) after malloc().

jolivepetrus, 2016-12-20:
In pthread programming first you have to initialise the variables in the pthread style, using the initializer for that variable. For conditions, you must do:

pthread_cond_t cond = PTHREAD_COND_INITIALIZER;

Once variable is initialise you can use it in pthread functions. If you do this somewhere in your code prior to call pthread_cond_signal the mutext will be created once.

edbek, 2016-12-20:
Ok.

But then you have to alter pthread_cond_init() function.
There is a situation when cond->mutex.sem ! = NULL, and then pthread_cond_init() function does nothing, and it should release the semaphore.
For example, the function introduction (where inside there pthread_cond_init)

pthread_cond_wait()
pthread_cond_timedwait()
pthread_cond_signal()

can not release the semaphore !

jolivepetrus, 2016-12-20:
Let me to make a review ....

jolivepetrus, 2016-12-21:
Please, take a look at:

https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/22d4b1d99842933ccbee5830edc7e9f50e8631d4

jolivepetrus, 2017-01-04:
Let me to close this issue. No feedback were provided.

closed 2017-01-04 by jolivepetrus