whitecatboard/Lua-RTOS-ESP32 #10: stat function, fat timestamp and few others...

loboris, 2016-12-25:
While working on some code I've needed **stat** function to return the file timestamp (fat fs) and I've found it is not working (not only for timestamp, but always returns error).

I've managed to make it work, maybe you are interested to include the changes:
- added **stat.c** to _components/lua_rtos/syscalls_
- made some changes, as in _changes.patch_

I've also changed few things to address the following issues:
- **os.ls()** : does not show the file time for files on fat fs
- **os.cd()** : after changing the working directory it is not possible to change it back to root

All changes are in attached zip.
[changes.zip](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/672252/changes.zip)

jolivepetrus, 2016-12-27:
Thanks for report this. We are reviewing your patch, and many things will be take in consideration.

Some comments about that ....

I think that changes in chdir / os.cd() are not required:

Lua RTOS has a minimal unix mount like-implementation. Out of the box spiffs are mounted on the root directory (/) and fat are mounted in /sd directory. This is done in sys/mount.c:

static const struct mountp mountps[] = {
#if USE_SPIFFS
#if USE_FAT
    {"/", "sd", "/sd", "spiffs", "fat"},
#endif
#endif
    {NULL, NULL, NULL, NULL, NULL}
};

This allows to mount a logical file system structure into a physical file system structure. For example, a reference to /tmp is translated to /spiffs/tmp, and a reference to /sd/file is translated to /fat/file. /spiffs and /fat are the names used by vfs for register the file systems.

Looking at the patch for chdir I think that you expect to go to the root directory doing os.cd("/spiffs"), but this is not correct, unless the root directory has a spiffs directory. In Lua RTOS doing an os.cd("/spiffs") changes the current working directory to the spiffs directory into the root directory, and if spiffs is not found an exception is raided:

nil	/spiffs: Not a directory	20

If you want to go to the root directory you must do os.cd("/").

If you want to go to the SD card root directory you must do od.cd("/sd"), this is an especial case, because /sd is  the mount-point for the FAT file system.

jolivepetrus, 2016-12-27:
stat.c should not be necessary ...

stat syscall is defined in esp-idf/newlib/syscall_table.c:

    ._stat_r = &esp_vfs_stat,

When calling to stat(...) esp_vfs_stat is calling, but for some reason vfs_fat_fstat is not called.

jolivepetrus, 2016-12-27:
Your contributions to the Lua RTOS project have been included in commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/25717a3f9ab3e0342367626a2147f49662f5bc97.

Please, test your code with this commit, to ensure it meets your needs.

loboris, 2016-12-28:
Thanks for the explanations. It is working now.

closed 2016-12-28 by loboris