whitecatboard/Lua-RTOS-ESP32 #13: hw i2c & sensor module

loboris, 2017-01-08:
As isp-idf now supports hw i2c, I've changed the i2c module.
Software i2c is removed and only the hw i2c is now used.
Only i2c master mode is supported for now.
The module is tested and works as expected.
I've also added the sensor module, for now support for BME280 sensor (in i2c mode) is added.
All related patches are in 
[i2c_sensor.patch.zip](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/692090/i2c_sensor.patch.zip)
You can also check my [Lua-RTOS-ESP32 fork](https://github.com/loboris/Lua-RTOS-ESP32)

Best regards.

jolivepetrus, 2017-01-10:
Hi loboris,

We will create a new branch for give support to i2c in Lua RTOS.  I took a look at your proposal for i2c driver and I have some comments about it:

* Lua platform folder will be disappear in the future

* Drivers in Lua RTOS are intended to have all the intelligence for allow simple Lua modules, and in the future for other programming languages.

* Drivers lock those resources they need via driver_lock function. For example, if a driver need 2 pins that pins are lock by the driver. If later, if another driver lock  resource locked previously by another driver an error is raised.

* Lua modules take care about errors raised by drivers and an exeption is generated to the user if a driver error is raised. Lock resources are important for us because they avoid programming errors from the blockly environment.

The ADC driver is a good example for know more about this.

loboris, 2017-01-11:
Thanks for your comments, especially about driver_lock function, I've somehow missed it.

jolivepetrus, 2017-01-11:
You have i2c driver and i2c Lua module in master branch see commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/bb446ad1a4d75995a4f5d0124151914f65cb6ab4
.
We have to write the wiki for this module, but you have an example in (https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/master/components/spiffs_image/image/examples/lua/i2c.lua).

For sensors we are thinking about a more elaborate strategy that fits better with our educational perpectives. Some ideas about that:

* Sensors are things that use drivers, so it would better to put them in another folder
* Is possible to have a Lua module that encapsulates generic sensor behavior, such as filters for reduce noise, ...
* We are think in something like:

   instance1 = sensor. BME280(i2c.I2C0, address, speed, sda, scl)
   instance1:read()
   ... 

   instance2 = sensor.TMP36(adc.ADC0, adc.ADC_CH6)
   instance2:read()
   ...

closed 2017-01-17 by loboris