whitecatboard/Lua-RTOS-ESP32 #14: error running with latest esp-idf

loboris, 2017-01-09:
After latest esp-idf commits (09 Jan) Lua-RTOS-ESP32 builds OK, but cannot run:

`Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
`

jolivepetrus, 2017-01-09:
Yes, we are working on it since this moorning.

Las commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/f3d7591151646a02a06194d7460340ba59afb90f solves this issue.

closed 2017-01-09 by loboris

jolivepetrus, 2017-01-09:
Last commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/5003de4475c0f4240a33e1a8b342a8978d20ed42 solves issues related to mkdir and ls commands for spiffs and fat.

jolivepetrus, 2017-01-09:
Work done in last commit (https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/480b6f35bab963654f19f80f527415c324b71788)