whitecatboard/Lua-RTOS-ESP32 #17: sensors module

loboris, 2017-01-15:
I've created the sensors module which (for now) has support for _BME280 temperature/humidity/pressure sensor_ and _1-wire devices_ (for now DS1820 temperature sensors family).
I've tried to write the drivers and modules according to your rules and I would appreciate If you could find some time to review the code and possible give some comments.

All patches are in the attached zip and in [GitHub repository](https://github.com/loboris/Lua-RTOS-ESP32-lobo)
[sensors.patch.zip](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/706947/sensors.patch.zip)

jolivepetrus, 2017-01-16:
Fine!

My comments about owire:

* Add in owire.h for avoid double inclussion and build / no build:

#ifndef _OWIRE_H_
#define _OWIRE_H_

#if USE_OWIRE

.... your code ....

#endif

#endif /* _OWIRE_H_ */

* Add in owire.c for build / no build:

#include "luartos.h"

#if USE_OWIRE

.... your code ....

#endif

* Add in luartos.h.

#define USE_OWIRE 1

==> now is set to 1, tipically is calculated from Kconfig options ...

* You are using esp-idf gpio driver functions. It's recommended to use Lua RTOS driver functions. In the future Lua RTOS will be take care about lock resources at gpio level. If you use only Lua RTOS drivers your driver will be ported to other plarforms without any problems.

* ets_delay_us ....

It's better to use alternative POSIX functions, such us delay / udelay that are available in Lua RTOS (sys/delay.h).

This is commit in the official Lua RTOS repository (see commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/fca04c1887cea9131e0724289a235d66fa9a9bb0).

About sensors module. We are working on in since last week. Maybeit should be better to have our e-mails for work in different things and don't duplicate efforts.

my contact: jolive@iberoxarxa.com

loboris, 2017-01-17:
Thanks for the comments.
I'm using esp-idf  functions because 1-wire timing is critical, I'll try Lua RTOS driver functions to see if there is any difference/problem.
I don't think there will be locking problems, the gpio is locked when owire interface is initialized (see
_components/lua_rtos/Lua/modules/sensors.c#lsensor_ow_init_ function in [my GitHub repo](https://github.com/loboris/Lua-RTOS-ESP32-lobo))

I've made some changes since I've posted _sensors.patch_. DS1820 is now fully tested with multiple sensors on the bus, BME280 is also fully tested. DS1820 fully supports parasite power mode, so that the sensor can be connected with only two wires. I will also add DHT-11/DHT-12 support.
I've addes WS2812 RGB LED driver and support for it in _led_ module.
Kcofig has config options for sensor and led modules

loboris@gmail.com

closed 2017-01-19 by loboris