whitecatboard/Lua-RTOS-ESP32 #18: SPIFFS component for ESP32 as a distinct repository/project?

nkolban, 2017-02-07:
Howdy,
I am a consumer of the SPIFFS code in your Lua project and am enjoying it very much.  I wonder if would be possible for you to create a distinct repository for your SPIFFS ESP32 component?  Ideally I'd like to be able to GIT clone it as a unit.  I haven't heard of any plans for anyone else to produce a SPIFFS implementation for ESP32 and yours definitely works just great.

jolivepetrus, 2017-02-13:
Hi nkolban,

This sounds really good!! Let us to schelude that (after the first release of our IDE).

bhgv, 2017-10-10:
yes, separate git repo for spiffs and normal directories and dir-ops

nkolban, 2017-10-10:
I'm not sure it is needed anymore.  When I look at the latest ESP32 EPS-IDF releases (the "OS" that runs on the ESP32) they now have a copy of SPIFFs baked into the environment.  I don't know the providence of it, but it seems to "be SPIFFS" and works just fine.

jolivepetrus, 2017-10-27:
We close this issue. Please reopen it if necessary.

closed 2017-10-27 by jolivepetrus