whitecatboard/Lua-RTOS-ESP32 #19: NodeMCU compatibility?

joysfera, 2017-02-13:
I just wanted to ask if you're aware of long time existing https://github.com/nodemcu/nodemcu-firmware/ project and if there is a good reason for starting another Lua interpreter for ESP.

While I appreciate it and admire that you're using a much newer Lua and you're based on RTOS from the start it's still a pity that the API of libraries is different so the potential Lua programs will not be compatible at all.

If there was a possibility of cooperating between the projects in order to maintain the API for user programs compatible it would be great.

Thanks.

jolivepetrus, 2017-02-13:
Hi joysfera,

Lua RTOS was born two years ago, after a careful study of the solutions available at that time, and one of that solutions was the NodeMCU project. Lua RTOS is one of the pieces of the Whitecat ecosystem, that is being developed by a team of engineers, educators and living lab designers, designed for build Internet Of Things networks in an easy way.

Our main goal is to cover the gap between education and real applications, and many of the Lua RTOS functions and APIS are designed for accomplish this goal.

The Lua RTOS key points are:

* Last Lua interpreter
* Lua RTOS can execute Lua functions in a Free RTOS thread, sharing all threads the same Lua context

thread.start(function()
....
end)

* An unified sensor API:

s1 = sensor.setup("DHT11", pio.GPIO4)
s1:read("temperature")

* A powerful online IDE (this will be available in a few weeks) with support for two program models: blocks or lua.

I'm sure that Lua has enough power for write modules for made possible the interoperation between Lua RTOS and NodeMCU APIs, and vice-versa.

closed 2017-02-13 by jolivepetrus

stormwatch, 2020-03-13:
Hello. I just bought the cheapest board I could find which happens to be NodeMCU ESP-32S
https://docs.zerynth.com/latest/official/board.zerynth.nodemcu_esp32/docs/index.html 

Will I be able to install Lua RTOS on it?

the0ne, 2020-03-13:
Most probably yes.
Looks like an ESP32 CoreBoard.
Try that 👍🏻

stormwatch, 2020-03-13:
Awesome! Thank you. It seems to work. At least I could interact with the Lua runtime, although I couldn't see those nice white kitten ears when connecting via picocom. Is there a mailing-list or forum or chat avaliable so I don't pollute this issue further?