whitecatboard/Lua-RTOS-ESP32 #22: Selecting LUA_RTOS_LORA_NODE_BAND_US915=y fails to compile.

jcwren, 2017-04-13:
This is with commit 799b155. Selecting LUA_RTOS_LORA_NODE_BAND_EU868=y succeeds.

```
CC node/lmic/aes.o
CC node/lmic/lmic.o
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.c: In function 'decodeFrame':
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.c:1126:17: warning: unused variable 'window' [-Wunused-variable]
     const char *window = (LMIC.txrxFlags & TXRX_DNW1) ? "RX1" : ((LMIC.txrxFlags & TXRX_DNW2) ? "RX2" : "Other");
                 ^
CC node/lmic/lmic_hal.o
CC node/lmic/lora_lmic.o
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:117:2: error: 'DR_SF12' undeclared here (not in a function)
  DR_SF12, DR_SF11, DR_SF10, DR_SF9, DR_SF8, DR_SF7, DR_SF7B, DR_FSK
  ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:117:11: error: 'DR_SF11' undeclared here (not in a function)
  DR_SF12, DR_SF11, DR_SF10, DR_SF9, DR_SF8, DR_SF7, DR_SF7B, DR_FSK
           ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:117:53: error: 'DR_SF7B' undeclared here (not in a function)
  DR_SF12, DR_SF11, DR_SF10, DR_SF9, DR_SF8, DR_SF7, DR_SF7B, DR_FSK
                                                     ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:117:62: error: 'DR_FSK' undeclared here (not in a function)
  DR_SF12, DR_SF11, DR_SF10, DR_SF9, DR_SF8, DR_SF7, DR_SF7B, DR_FSK
                                                              ^
In file included from /Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:56:0:
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c: In function 'lora_init':
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.h:270:47: error: invalid operands to binary << (have 'int' and 'const u1_t * {aka const unsigned char *}')
 #define DR_RANGE_MAP(drlo,drhi) (((u2_t)0xFFFF<<(drlo)) & ((u2_t)0xFFFF>>(15-(drhi))))
                                               ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:218:38: note: in expansion of macro 'DR_RANGE_MAP'
      LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);
                                      ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.h:270:57: error: invalid operands to binary & (have 'const u1_t * {aka const unsigned char *}' and 'int')
 #define DR_RANGE_MAP(drlo,drhi) (((u2_t)0xFFFF<<(drlo)) & ((u2_t)0xFFFF>>(15-(drhi))))
                                                         ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:218:38: note: in expansion of macro 'DR_RANGE_MAP'
      LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);
                                      ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:218:70: error: 'BAND_CENTI' undeclared (first use in this function)
      LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);
                                                                      ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:218:70: note: each undeclared identifier is reported only once for each function it appears in
In file included from /Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:56:0:
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.h:270:33: warning: passing argument 3 of 'LMIC_setupChannel' makes integer from pointer without a cast [-Wint-conversion]
 #define DR_RANGE_MAP(drlo,drhi) (((u2_t)0xFFFF<<(drlo)) & ((u2_t)0xFFFF>>(15-(drhi))))
                                 ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:218:38: note: in expansion of macro 'DR_RANGE_MAP'
      LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);
                                      ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.h:275:7: note: expected 'u2_t {aka short unsigned int}' but argument is of type 'const u1_t * {aka const unsigned char *}'
 bit_t LMIC_setupChannel (u1_t channel, u4_t freq, u2_t drmap, s1_t band);
       ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:218:70: warning: passing argument 4 of 'LMIC_setupChannel' makes integer from pointer without a cast [-Wint-conversion]
      LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);
                                                                      ^
In file included from /Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:56:0:
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.h:275:7: note: expected 's1_t {aka signed char}' but argument is of type 'const u1_t * {aka const unsigned char *}'
 bit_t LMIC_setupChannel (u1_t channel, u4_t freq, u2_t drmap, s1_t band);
       ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.h:270:47: error: invalid operands to binary << (have 'int' and 'const u1_t * {aka const unsigned char *}')
 #define DR_RANGE_MAP(drlo,drhi) (((u2_t)0xFFFF<<(drlo)) & ((u2_t)0xFFFF>>(15-(drhi))))
                                               ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:219:38: note: in expansion of macro 'DR_RANGE_MAP'
      LMIC_setupChannel(1, 868300000, DR_RANGE_MAP(DR_SF12, DR_SF7B), BAND_CENTI);
                                      ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.h:270:77: error: invalid operands to binary - (have 'int' and 'const u1_t * {aka const unsigned char *}')
 #define DR_RANGE_MAP(drlo,drhi) (((u2_t)0xFFFF<<(drlo)) & ((u2_t)0xFFFF>>(15-(drhi))))
                                                                             ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:219:38: note: in expansion of macro 'DR_RANGE_MAP'
      LMIC_setupChannel(1, 868300000, DR_RANGE_MAP(DR_SF12, DR_SF7B), BAND_CENTI);
                                      ^
[... 222 lines cut ...]
                                                                      ^
In file included from /Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:56:0:
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.h:270:33: warning: passing argument 3 of 'LMIC_setupChannel' makes integer from pointer without a cast [-Wint-conversion]
 #define DR_RANGE_MAP(drlo,drhi) (((u2_t)0xFFFF<<(drlo)) & ((u2_t)0xFFFF>>(15-(drhi))))
                                 ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:226:38: note: in expansion of macro 'DR_RANGE_MAP'
      LMIC_setupChannel(8, 868800000, DR_RANGE_MAP(DR_FSK,  DR_FSK),  BAND_MILLI); 
                                      ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.h:275:7: note: expected 'u2_t {aka short unsigned int}' but argument is of type 'const u1_t * {aka const unsigned char *}'
 bit_t LMIC_setupChannel (u1_t channel, u4_t freq, u2_t drmap, s1_t band);
       ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:226:70: warning: passing argument 4 of 'LMIC_setupChannel' makes integer from pointer without a cast [-Wint-conversion]
      LMIC_setupChannel(8, 868800000, DR_RANGE_MAP(DR_FSK,  DR_FSK),  BAND_MILLI); 
                                                                      ^
In file included from /Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:56:0:
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lmic.h:275:7: note: expected 's1_t {aka signed char}' but argument is of type 'const u1_t * {aka const unsigned char *}'
 bit_t LMIC_setupChannel (u1_t channel, u4_t freq, u2_t drmap, s1_t band);
       ^
/Users/jcw/esp/Lua-RTOS-ESP32/components/lora/./node/lmic/lora_lmic.c:240:16: warning: assignment makes integer from pointer without a cast [-Wint-conversion]
     current_dr = DR_SF12;
                ^
make[1]: *** [node/lmic/lora_lmic.o] Error 1
make: *** [lora-build] Error 2
mbpro:Lua-RTOS-ESP32 $
```

jolivepetrus, 2017-04-18:
@jcwren, many thanks for report this.

The compilation issue is solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/0ee3fa89f7dee733835d1615ae72f83d88ce36c4.

This commit solves the issue, but it's not tested because we can't use this band in our location. Please send us your feedback.

jcwren, 2017-04-23:
I've gotten it to compile, but having an issue trying to pass BAND915 to lora.setup(). 

Do you need any additional information?

```
  /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1 build 1492979468 Copyright (C) 2015 - 2017 whitecatboard.org
board type ESP32THING
cpu ESP32 rev 0 at 240 Mhz
spiffs0 start address at 0x180000, size 512 Kb
spiffs0 mounted

Lua RTOS beta 0.1 powered by Lua 5.3.4

Executing /system.lua ...
Executing /autorun.lua ...

/ > lora.setup(lora.BAND868)
invalid band for your location
stdin:1: 100663307:invalid band for your location
stack traceback:
	[C]: in field 'setup'
	stdin:1: in main chunk
	[C]: in ?
/ > 
lora.setup(lora.BAND915)
stdin:2: 100663305:invalid band
stack traceback:
	[C]: in field 'setup'
	stdin:2: in main chunk
	[C]: in ?
/ > 
lora.setup(lora.BAND433)
invalid band for your location
stdin:2: 100663307:invalid band for your location
stack traceback:
	[C]: in field 'setup'
	stdin:2: in main chunk
	[C]: in ?
/ > 
lora.setup(lora.BAND123)
stdin:2: bad argument #1 to 'setup' (number expected, got nil)
stack traceback:
	[C]: in field 'setup'
	stdin:2: in main chunk
	[C]: in ?
/ >
```

jolivepetrus, 2017-04-24:
Please, attach your sdkconfig file.

jcwren, 2017-04-24:
I can do that when I get home this evening. However, the problem line is ```if ((band != 868) && (band != 433)) {``` in ./components/lua_rtos/Lua/module/lora.c, where it's not permitting 915 as a valid value.

jolivepetrus, 2017-04-24:
Yes, you are right. Please, try with this commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/e330d2ebe87490991f8cb457cd3d81430472f614.

jolivepetrus, 2017-05-09:
Let me to close this issue. No feedback were provided.

closed 2017-05-09 by jolivepetrus

jcwren, 2017-05-10:
Sorry about that. I've had to temporarily preempt this project and haven't been able to test it. If I find an issue, I'll re-open it.  Thanks.