whitecatboard/Lua-RTOS-ESP32 #23: MQTT does not support encrypted transmission

the0ne, 2017-04-26:
The current MQTT implementation relys on OpenSSL being available at compile time.
The Problem is that OpenSSL is not used for the creation of the lua_rtos binary but mbedTLS is being used.
For this reason the MQTT implementation needs to be ported to mbedTLS as unencrypted MQTT is not an option except when using a local MQTT broker because of MITM-Attacks, Replay Attacks, etc.

jolivepetrus, 2017-04-27:
@the0ne, thanks for report this.

In the coming weeks a public mqtt server will be available at whitecatboard.org, where users who are evaluating Lua RTOS will be able to perform mqtt tests in both plain and encrypted transmissions.

the0ne, 2017-04-28:
If it is of any help I can offer access to a broker that is set up already, reachable 24/7 and offering encrypted authenticated connections.

jolivepetrus, 2017-04-28:
Perfect!

Yes, this would be a great help to us. You can contact me at admin@whitecatboard.org for more details.

the0ne, 2017-05-10:
I managed to fix several issues and to "port" the SSLSocket code to the mbedTLS openssl abstraction layer. So basically encrypted MQTT works, though not with all security features like client certs or full server validation. There's some more issues that need to be checked, tough - e.g. client ids when using several threads.

jolivepetrus, 2017-05-10:
Gonna look into it, but at the moment I'm slightly distracted by The Whitecat IDE.

Can you make a pull request for test it and vell you to completes the work?

the0ne, 2017-05-15:
see #30 commit bfaca2f6c818fe54673da22c07d6b9c7fc659061

closed 2017-05-16 by the0ne