whitecatboard/Lua-RTOS-ESP32 #24: HTTP Server static only

the0ne, 2017-04-26:
The HTTP Server does not seem to interpret Lua code which seems like a show-stopper.
Is there a way to workaround this limitation?

jolivepetrus, 2017-04-27:
@the0ne, thanks for report this.

HTTP Server does not interpret Lua code for now. Now we are migrating the lua sockets api (http://w3.impa.br/~diego/software/luasocket). A more powerful http server will be planned later.

the0ne, 2017-05-11:
I have added lua interpreter calls to the webserver and a basic http lua module 'common-http.lua' to the spiffs file system.

Now i can execute lua scripts by calling e.g.
http://_esp32-client-ip_/examples/lua/http-test.lua?ssid=_ssid-here_&pass=_pass-here_
and the file 'http-test.lua' is as short as:
> require('common-http')
> get = parseurl(http_request)
> http_response = ""
>
> print(get.ssid)
> print(get.pass)

The overridden print function makes sure the output is not sent to the console but to the http client.

jolivepetrus, 2017-05-15:
Can you create a pull request with your code?

:)

the0ne, 2017-05-15:
Sure, i promise :-)
I had planned to do this already but didn't find the time.
btw: I have started work on the wifi AP mode.

jolivepetrus, 2017-05-15:
Fantasic!!

the0ne, 2017-05-15:
see #30 commit c4b058771be1b935a245afe59983282c7aeaf7e9

jolivepetrus, 2017-05-16:
We have merged this commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/c4b058771be1b935a245afe59983282c7aeaf7e9 because is not a function currently used.
The proposed implementation has some problems:

* It uses a global variable for return the generated code. This approach is not thread-safe.
* In common-http.lua the print function is redefined. After serving the first lua file other threads, don't see the original print function.
* Httpsrv implementation was an experiment for us, but is a poor implementation, with very few capabilities.

I was thinking about something like this:

<html>
..
..
<? print("I'm Lua") ?>
..
..
</html>

I think that a deep design of this functionality is required.

the0ne, 2017-05-16:
I totally agree - that will be perfect.

jolivepetrus, 2017-05-16:
In a few days I expect to start the work on the Lua sockets API. Probably is the start point for that.

closed 2017-05-16 by the0ne