whitecatboard/Lua-RTOS-ESP32 #25: Australian Lorawan Frequencies

MrAPierce, 2017-04-28:
Hi, great board and nice code, would it be possible to support the Australian and New Zealand lorawan frequencies? I see that you are using the arduino-lmic code which has defined configs such as CFG_us915/CFG_eu868 which sets up the lorawan related functions ... could other configs could be added? would be great if you could support more than just US and EU out of the box.

Cheers

jolivepetrus, 2017-04-29:
More regions, including Australia, will be supported soon. Since some weeks we are working in the Semtech stack.

jolivepetrus, 2017-05-02:
A new branch have been created for publish initial work in the semtech stack: https://github.com/whitecatboard/Lua-RTOS-ESP32/tree/semtech-stack.

This branch is for development purposes only and may not work if you cloned it.