whitecatboard/Lua-RTOS-ESP32 #26: Using uart1 or uart2 does not work

JanHouwers, 2017-05-01:
Hello,
I am trying to read serial data from a PMS1003 dust sensor. The dust sensor works correct and i receive valid data when using a 3.3v serial to USB convertor. The sensor sends every 2 seconds a 32 byte string.
When connecting the sensor to port 9 and 10 (uart1) no valid data is returned to the LUA program.
When connecting a 3.3v serial to USB converter instead of the sensor only the first byte is returned until the queue is empty.

Also when using: 
 > print (uart.read(uart.UART1, "*l", 1000))
you get:
stdin:1: bad argument #3 to 'read' (boolean expected, got number)
stack traceback:
	[C]: in field 'read'
	stdin:1: in main chunk
	[C]: in ?
It does not matter if anything is received first. Also when using:
 > print (uart.read(uart.UART1, "*c", 1000))

Nothing valid returns.. 

Regards, Jan

jolivepetrus, 2017-05-02:
@JanHouwers, thanks for report this.

I think that problem is caused by UART pin maps. We have done some UART driver refactory that allows to customize the UART pin maps.

Pin maps for WHITECAT N1, ESP32 THING, and ESP32 CORE, boards have been updated:

WHITECAT N1:
===========
RX0: GPIO3
TX0: GPIO1
RX1: GPIO26
TX1: GPIO25
RX2: GPIO2
TX2: GPIO15

ESP32 THING:
===========
RX0: GPIO3
TX0: GPIO1
RX1: GPIO12
TX1: GPIO13
RX2: GPIO16
TX2: GPIO17

CORE BOARD:
===========
RX0: GPIO3
TX0: GPIO1
RX1: GPIO12
TX1: GPIO13
RX2: GPIO2
TX2: GPIO17

Please, try again with this commit: https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/da240ae9090667c8ba6144567ec425c9e3d6e4b2.

JanHouwers, 2017-05-02:
Hello,

Just tested with the Whitecatboard and i can read the data from the PMS1003 sensor in a reliable manner..

Regards, Jan

closed 2017-05-09 by jolivepetrus