whitecatboard/Lua-RTOS-ESP32 #29: Latest version (703abe1) errors when building CAN.c

jcwren, 2017-05-13:
Pulled a fresh copy of esp-idf, xtensa-esp32-elf, Lua-RTOS-ESP32. Changed board to ESP32 THING, no other modifications to configuration. Fails when building CAN.c. Configuration at
[sdkconfig.txt](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/998669/sdkconfig.txt)

```
$ make
cc  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o mconf.o mconf.c
flex -L -P zconf -o zconf.lex.c zconf.l
zconf.l:255: warning, -s option given but default rule can be matched
bison -t -l -p zconf -o zconf.tab.c zconf.y
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
cc  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o zconf.tab.o zconf.tab.c
lxdialog/check-lxdialog.sh -check cc  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -lncurses 
cc  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/checklist.o lxdialog/checklist.c
cc  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/util.o lxdialog/util.c
cc  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/inputbox.o lxdialog/inputbox.c
cc  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/textbox.o lxdialog/textbox.c
cc  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/yesno.o lxdialog/yesno.c
cc  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/menubox.o lxdialog/menubox.c
cc -o mconf mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o -lncurses 
cc  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o conf.o conf.c
cc -o conf conf.o  zconf.tab.o -lncurses 
GENCONFIG
Configure esp-idf ...
Configure esp-idf for Lua RTOS ...
patching file /Users/jcw/Projects/esp-idf/components/lwip/api/api_msg.c
Hunk #1 succeeded at 875 (offset 6 lines).
patching file /Users/jcw/Projects/esp-idf/components/vfs/include/sys/dirent.h
patching file /Users/jcw/Projects/esp-idf/components/spi_flash/flash_ops.c
Hunk #1 succeeded at 486 with fuzz 1 (offset 7 lines).
patching file /Users/jcw/Projects/esp-idf/components/spi_flash/include/esp_spi_flash.h
Hunk #1 succeeded at 347 (offset 42 lines).
patching file components/esp32/event_default_handlers.c
patching file components/esp32/include/esp_event.h
patching file components/esp32/include/esp_interface.h
patching file components/tcpip_adapter/include/tcpip_adapter.h
patching file components/tcpip_adapter/tcpip_adapter_lwip.c
Hunk #3 succeeded at 110 (offset 1 line).
Hunk #4 succeeded at 178 (offset 1 line).
Hunk #5 succeeded at 207 (offset 1 line).
Hunk #6 succeeded at 242 (offset 1 line).
Hunk #7 succeeded at 265 (offset 1 line).
Hunk #8 succeeded at 326 (offset 1 line).
Hunk #9 succeeded at 341 (offset 1 line).
Hunk #10 succeeded at 443 (offset 1 line).
Hunk #11 succeeded at 461 (offset 1 line).
Hunk #12 succeeded at 664 (offset 1 line).
Hunk #13 succeeded at 675 (offset 1 line).
Hunk #14 succeeded at 694 (offset 1 line).
Hunk #15 succeeded at 721 (offset 1 line).
Hunk #16 succeeded at 845 (offset 1 line).
Configure esp-idf for Lua RTOS tests ...
CC src/bootloader_flash.o
<command-line>:0:0: warning: "IDF_VER" redefined
<command-line>:0:0: note: this is the location of the previous definition
CC src/bootloader_random.o
<command-line>:0:0: warning: "IDF_VER" redefined
<command-line>:0:0: note: this is the location of the previous definition
CC src/efuse.o
<command-line>:0:0: warning: "IDF_VER" redefined
<command-line>:0:0: note: this is the location of the previous definition
CC src/esp_image_format.o
<command-line>:0:0: warning: "IDF_VER" redefined
<command-line>:0:0: note: this is the location of the previous definition
CC src/flash_encrypt.o
<command-line>:0:0: warning: "IDF_VER" redefined
<command-line>:0:0: note: this is the location of the previous definition
CC src/flash_partitions.o
<command-line>:0:0: warning: "IDF_VER" redefined
<command-line>:0:0: note: this is the location of the previous definition
CC src/secure_boot.o
[... 172 lines cut ...]
/Users/jcw/Projects/esp-idf/components/soc/esp32/include/soc/soc.h:211:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((addr), READ_PERI_REG);                                                                \
             ^
/Users/jcw/Projects/esp-idf/components/soc/esp32/include/soc/soc.h:224:36: note: in expansion of macro 'READ_PERI_REG'
             WRITE_PERI_REG((reg), (READ_PERI_REG(reg)&(~(mask))));                                                     \
                                    ^
/Users/jcw/Projects/Lua-RTOS-ESP32/components/can/./CAN.c:148:5: note: in expansion of macro 'CLEAR_PERI_REG_MASK'
     CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_CAN_RST);
     ^
make[1]: *** [CAN.o] Error 1
make: *** [can-build] Error 2
$
```
Compiler version:
```
$ xtensa-esp32-elf-gcc -v
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/Users/jcw/Projects/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /Users/ivan/e/ESPTools/crosstool-NG/.build/src/gcc-5.2.0/configure --build=x86_64-build_apple-darwin16.0.0 --host=x86_64-build_apple-darwin16.0.0 --target=xtensa-esp32-elf --prefix=/Users/ivan/e/ESPTools/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/Users/ivan/e/ESPTools/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-sysroot=/Users/ivan/e/ESPTools/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-61-gab8375a' --disable-__cxa_atexit --enable-cxx-flags='-fno-exceptions -fno-rtti' --with-gmp=/Users/ivan/e/ESPTools/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/Users/ivan/e/ESPTools/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/Users/ivan/e/ESPTools/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/Users/ivan/e/ESPTools/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-cloog=/Users/ivan/e/ESPTools/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-libelf=/Users/ivan/e/ESPTools/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-nls --disable-multilib --enable-languages=c,c++ --disable-libstdcxx-verbose
Thread model: single
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 
$
```

jolivepetrus, 2017-05-15:
Yes, fails in all platforms and supported boards, due to a recent esp-idf update.
We are working on it.

jolivepetrus, 2017-05-15:
This build error is caused by esp-idf commit https://github.com/espressif/esp-idf/commit/f7e88565208451a7c3415be3a2b94d32377441f6#diff-0e0de619292959fa4d943440d4a6f161.

jolivepetrus, 2017-05-15:
Please, try with this commit: https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/f98b7cd337e36b9e9708496d314936a210ceee96

After changes CAN driver builds without any problem.

Tested on Mac OSX.

I can't test now with CAN hardware, but changes are only related to enable / disable hardware CAN module with new macros proposed by Espressif.

closed 2017-05-15 by jolivepetrus

jcwren, 2017-05-15:
Excellent, I am also able to build without errors. I have no way to test the actual CAN functionality, but I was unable to build regardless if CAN was enabled or disabled via 'make menuconfig'. Thanks!