whitecatboard/Lua-RTOS-ESP32 #39: Add method to disable SPI2 or SPI3 when SPI enabled?

jcwren, 2017-05-21:
Currently, if `CONFIG_LUA_RTOS_LUA_USE_SPI` == y, both SPI2 and SPI3 are enabled and the respective GPIO pins are configured. Given that the I/O on the ESP32 is somewhat limited, it would be advantageous to be able to selectively enable SPI2 or SPI3.

This could be done by either having a sub-menu in the SPI configuration and allowing SPI2 and SPI3 to be selectively enabled, or by allowing the pins in the SPI pin map menu to be set to -1.

Additionally, it would be helpful if the SPI TX or RX pin on each SPI channel could be selectively disabled. There are some devices that don't need to read data back, and being able to disable the RX pin and leave it available as a GPIO might be helpful.

Thoughts?

jolivepetrus, 2017-05-22:
In Kconfig only SPI PIN map is configured.
SPI pins are only configured in spi_setup_bus which is called from spi_ll_setup, so SPI pins are only configured when you setup SPI from lua, using the spi.setup function.

If you only use SPI1, for example, only SPI1 pins are configured.

jcwren, 2017-05-22:
Got it. I'd still like to have a way to selectively disable MISO or MOSI to keep the GPIO pin available, but I see what you're saying about the SPI GPIO not actually being configured until the spi.setup() call.

If you're agreeable to selectively disabling MISO and MOSI we can leave this open, otherwise mark it closed.

jolivepetrus, 2017-05-22:
Yes, selectively disable MISO / MOSI in SPI driver might be helpful.

This will be treated in a new issue: https://github.com/whitecatboard/Lua-RTOS-ESP32/issues/42

closed 2017-05-22 by jolivepetrus