whitecatboard/Lua-RTOS-ESP32 #42: Selectively disable MISO or MOSI in SPI driver

jolivepetrus, 2017-05-22:
It would be helpful if the SPI MOSI or MISO pin on each SPI channel could be selectively disabled. There are some devices that don't need to read data back, and being able to disable the MISO pin and leave it available as a GPIO might be helpful.

jolivepetrus, 2017-05-22:
We have to think about that. Keep in mind that Lua RTOS it's a very important part of the Whitecat ecosystem, that can be programmed in blocks or lua thought The Whitecat IDE, and this enhancement affects to the block programming model.

jolivepetrus, 2017-05-22:
A proposal would be to enable MISO in SPI reads (only once) and MOSI in SPI writes (only once). This solution is transparent to the programmer, and no changes to the API are required, even is possible to attach in the same SPI bus a device that only use MISO, and other that only uses MOSI.

jolivepetrus, 2017-05-22:
The same can be applied to the UART (for example in GPS).

jcwren, 2017-05-22:
I'm not sure there's a use case where having two devices on a SPI bus, one of which uses only MISO and one of which uses only MOSI would be useful, as the GPIO pins couldn't be practically used for other things.

In the majority of cases where this would be useful is when a single device is on the SPI bus. For this case, it would sufficient to just set the pin map variable (such as `CONFIG_LUA_RTOS_SPI2_MISO`) to -1.

Alternatively, but a little more complicated, would be to add a mode to the SPI set up, such as MODE_RXONLY, MODE_TXONLY, and MODE_TXRX. That would seem to have a wider ranging impact, as it would affect the API.

Good point on the UART.

jolivepetrus, 2017-05-22:
Second option seems to be more flexible.

SPI: add a new argument "flags" in setup function, with values SPI.Write = 0b01, SPI.Read = 0b10, with default value SPI.write | SPI.read. In the future can be used for other things, such us configure full duplex / no full duplex.

UART: add a new argument "flags" in setup function, with values UART.Write = 0b01, UART.Read = 0b10, with default value UART.Write | UART.Read.

jolivepetrus, 2017-05-24:
Solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/3e0699e8a5e759ccc3853de892a2c09de55d453c.

Please, try if you can.

jcwren, 2017-05-24:
I will try to do so by the weekend. I'm waiting for a couple components to arrive from Mouser to finish building up my prototype. I have a few LoRa devices on the air already (RN2903, Feather M0 RMF95W, and some LinkLabs kit), and want to get the ESP32 Thing up also.

jolivepetrus, 2017-05-24:
Same for UART is implemented in commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/8842a75f30d2e9974b5a51b4d3773194389e82f2.

closed 2017-06-12 by jolivepetrus