whitecatboard/Lua-RTOS-ESP32 #43: Guru Mediaton Error if readonly cache enabled and table element doesn't exists

jolivepetrus, 2017-05-22:
If "Use cache for readonly tables access (experimental)" is enabled in Kconfig, if a table's key doesn't exists a Guru Mediaton Error is raided. Examles:

/ > lora.inexistent
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4011a2af  PS      : 0x00060c30  A0      : 0x8010e258  A1      : 0x3ffc7120  
A2      : 0x3ffc770c  A3      : 0x3ffc9f9c  A4      : 0x3ffc8d40  A5      : 0x3ffca174  
A6      : 0x3ffc9f94  A7      : 0x3ffca174  A8      : 0x8011a10c  A9      : 0x3ffc70f0  
A10     : 0x00000000  A11     : 0x3ffc9f9c  A12     : 0x00000101  A13     : 0x00000000  
A14     : 0x0000002a  A15     : 0x3ffc7290  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

the0ne, 2017-05-24:
Seeing the PC here reminds me of two scripts I created recently.
On my PC I have put both into ../xtensa-esp32-elf/bin/

backtrace.sh
> #!/bin/bash
>
>if [ -z $1 ] ; then
> 	echo -n "backtrace: "
> 	read backtrace
>else
> 	backtrace=$*
>fi
>
>BASE=$(dirname $0)
>for address in $backtrace; do $BASE/getline.sh $address; done

getline.sh

> #!/bin/bash
>
> if [ -z $1 ] ; then
>	echo -n "address: "
>	read address
>else
>	address=$1
>fi
>
>xtensa-esp32-elf-addr2line -aipsfC -e ./build/lua_rtos.elf $address

So I just need to call 
> ~/esp32/Lua-RTOS-ESP32$ backtrace.sh 0x40001d53:0x3ffcbad0 0x401152c8:0x3ffcbaf0 0x40132702:0x3ffcbb10 0x4010479d:0x3ffcbb30 0x40115673:0x3ffcbb50 0x4011580d:0x3ffcbb70 0x40115834:0x3ffcbb90 0x40115c8c:0x3ffcbbb0 0x401152d6:0x3ffcbbd0 0x4011598d:0x3ffcbc50 0x40116925:0x3ffcbc80 0x400fbd7e:0x3ffcbcb0 0x4010fdb4:0x3ffcbe70 0x4010ff06:0x3ffcbff0 0x40110125:0x3ffcc010 0x40110199:0x3ffcc030 0x40115673:0x3ffcc050 0x4011580d:0x3ffcc070 0x40115834:0x3ffcc090 0x40115c8c:0x3ffcc0b0 0x401152d6:0x3ffcc0d0 0x4011598d:0x3ffcc150 0x40116925:0x3ffcc180 0x401101d3:0x3ffcc1b0 0x401447b6:0x3ffcc1d0 0x400fb79d:0x3ffcc1f0

Where all the numbers are just copy/paste out of the gtkterm window.

jolivepetrus, 2017-05-26:
Solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/51fab13fa5867722d01cf9c6fb7423ab7084a37a.

jcwren, 2017-05-28:
"They're baaaaaack"...  This is with the version just prior to committing the thread check change.

[screenlog.txt](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/1034561/screenlog.txt)
[backtrace.txt](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/1034562/backtrace.txt)
[sdkconfig.txt](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/1034564/sdkconfig.txt)

jolivepetrus, 2017-05-28:
Try with https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/204a01fed97ef8afc522c56b9e1cda56edc0da8a.

jcwren, 2017-05-28:
That fixes it, and works with CONFIG_LUA_RTOS_LUA_USE_ROTABLE_CACHE both set and unset.

closed 2017-05-28 by jolivepetrus