whitecatboard/Lua-RTOS-ESP32 #49: I2S support -- philosophical question how to handle data formats

jcwren, 2017-05-24:
I'm in the process of adding I2S support, and hitting an issue about how to best pass the data back and forth between the C API and Lua.

When writing to the I2S device, the Espressif driver wants a buffer of type `const char *`. Should the Lua write function be passed a string or userdata memory? A string seems unnatural for manipulating audio data, but the userdata would require providing a series of methods for modifying the memory from Lua.

The same holds true for the read(), push(), and pop() functions. Should the data being returned be strings or userdata? And for the read() and pop() functions, should the user provide a pre-allocated buffer, or should I dynamically allocate these? There can be a lot of data flying back and forth in I2S, so static buffers would appear to be more efficient.

If you'd like to look at what I've done so far (and optionally make fun of it), it's at [https://github.com/jcwren/Lua-RTOS-ESP32/tree/i2s_support](https://github.com/jcwren/Lua-RTOS-ESP32/tree/i2s_support). While it does compile cleanly, it's still not functional (or even tested on real hardware yet).

jolivepetrus, 2017-05-24:
Historically, in Lua, this is done using strings and the pack / unpack functions from the string module. In Lua RTOS this is done in the UART and SPI module.

In Lua RTOS user data is used in cases where a setup function creates an instance, for example in the sensor module,

Another approach is the used in the Lora WAN module, where payload (binary data) interchange is done using hex-strings, for example: 01AB02 is 0x01ab02.

For coherence I thinks that the pack / unpack described above is better.

I had a look on your i2s code and it's nice!! Congratulations.

jcwren, 2017-05-28:
OK, I'll go with the pack/unpack method. I've got everything written, now I just need to get past this latest endless reboot issue (#52) so I can test it.

jolivepetrus, 2017-10-27:
We close this issue. Please reopen it if necessary.

closed 2017-10-27 by jolivepetrus