whitecatboard/Lua-RTOS-ESP32 #50: problem compiling using msys32 with error  unrecognized command -mno-ms-bitfields

jevers76, 2017-05-26:
While trying to compile Lua-RTOS_ESP32 I am getting the error:
xtensa-esp32-elf-gcc.exe: error: unrecognized command line option '-mno-ms-bitfields'
It seem like the $(OS) used is Windows_NT instead of LINUX perhaps or MSYS32.  Is this correct? The LUA-RTOS-32, esp-idf and MSYS are the newest versions in github.   Below is a screen capture:

john@hpi7 MSYS ~/Lua-RTOS-ESP32
$ make flashfs
GENCONFIG
Building mkspiffs ...
xtensa-esp32-elf-gcc.exe: error: unrecognized command line option '-mno-ms-bitfields'
make[1]: *** [Makefile:58: mkspiffs.exe] Error 1
make: *** [/home/john/Lua-RTOS-ESP32/components/mkspiffs/Makefile.projbuild:10: mkspiffs] Error 2

jolivepetrus, 2017-05-26:
Something is wrong in the toolchain!

make flashfs, builds the mkspiffs tool for build and flash the spiffs image. This tool must be compiled with the host's toolchain, because it's executed in the host. I can see that for some reason in your installation the esp32 toolchain is used.

I will try to reproduce it at home this weekend.

jolivepetrus, 2017-05-26:
A similar issue was solved (Ubuntu) in this commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/7663bac363c418be7153b530a42feabc2fcba07c.

jolivepetrus, 2017-10-27:
We close this issue. Please reopen it if necessary.

closed 2017-10-27 by jolivepetrus