whitecatboard/Lua-RTOS-ESP32 #52: Latest build reboots endlessly with Guru Meditation Error

jcwren, 2017-05-28:
Meditation error needs medication :)

Bone-stock build using Lua-RTOS-ESP32 51fab13 and esp-idf 636d2b3, configured for ESP32 Thing.

[sdkconfig.txt](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/1034347/sdkconfig.txt)
[screenlog.txt](https://github.com/whitecatboard/Lua-RTOS-ESP32/files/1034345/screenlog.txt)

jolivepetrus, 2017-05-28:
No problem if compiled with proposed sdkconfig

make SDKCONFIG_DEFAULTS=ESP32-THING defconfig

Trying to find the cause ....

jcwren, 2017-05-28:
Reasonably sure it's because of the latest esp-idf updates. I was booting the image last night, updated esp-idf today ('git pull', 'git submodule update'), and that's when it lost it's mind.

jolivepetrus, 2017-05-28:
The problem is that in your sdkconfig CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=1, when the proposed sdkconfig for ESP32THING is CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=2. I'll add some check for avoid this, because LUARTOS needs an especific local storage.

jcwren, 2017-05-28:
Did something change that selected that? I pulled fresh copies, did a 'make menuconfig', set the board type, exited w/ save, then compiled. That's not been an issue in the past.

jolivepetrus, 2017-05-28:
esp-idf sets this to 1.

Prior to compile Lua RTOS do:

rm sdkconfig
rm sdkconfig.old
make SDKCONFIG_DEFAULTS=ESP32-THING defconfig

In the default config for ESP32THING and otheers, CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS is set to 2.

jcwren, 2017-05-28:
Fair enough. If I remove the sdkconfig files, run 'make menuconfig' and set the board type, then CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=1. I would have expected that to generate the same results as running 'make SDKCONFIG_DEFAULTS=ESP32-THING defconfig' since there is no sdkconfig file present. 

Interesting. If I diff the sdkconfig's generated by 'make menuconfig' and 'make SDKCONFIG_DEFAULTS=ESP32-THING defconfig' there are a fair number of differences, not something I would have expected.

I'll make sure to use the command line version as my base starting point.

jolivepetrus, 2017-05-28:
We build the default configurations from the default's esp-idf: ESP32-CORE-BOARD, ESP32-THING, GENERIC, WHITECAT-ESP32-G1, WHITECAT-ESP32-N1. The default configs for Lua RTOS differs from the originals.

For example, in the Whitecat N1, the Lora WAN transceiver is enabled by default, but in ESP32THING this option is disabled.

There are more changes, such as the flash speed, dio / qio, stack sizes, etc ...

jcwren, 2017-05-28:
OK, that fixed it, it's running again. I wonder why I haven't been bitten by that before, since I have wiped out the sdkconfig files in the past.

I don't suppose it's possible to configure the menuconfig files to force reset everything if you change the board type?

jolivepetrus, 2017-05-28:
No, it is not possible, at least as far as I know.

jolivepetrus, 2017-05-28:
A minimal check for this issue is implemented in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/9351e36c886cac295d00d8f57df20a789a13fba8.

closed 2017-05-28 by jcwren