whitecatboard/Lua-RTOS-ESP32 #53: LoRaWAN Reboot

webfrank, 2017-06-09:
Hi, I wrote a simple script which sends a sample data:
```
-- ----------------------------------------------------------------
-- Send a frame via LoRa WAN using ABP
-- ----------------------------------------------------------------
os.loglevel(os.LOG_ALL)

-- Register received callbacks
lora.whenReceived(function(port, payload)
  print("port "..port)
  print("payload "..payload)
end)

thread.start(function()
-- Setup LoRa WAN
    lora.setup(lora.BAND868)
    lora.setDevAddr("020FD5D2")
    lora.setNwksKey("BFDE6748451AE5FAC129F579EA4B97B1")
    lora.setAppsKey("6987FB08D1C5447F49591D5AE011C135")
    lora.setDr(5)
    lora.setReTx(0)

    while true do
        print("Sending data...")
        -- Send frame
        lora.tx(false,1,pack.pack(25.3))
        
        -- Wait 5 seconds
        tmr.delay(15)
    end
end)
```

But the boards (ESP32CORE with RFM96 SPI connected) reboots:

```
dofile('/lora.lua')
/ > lora: setup, band 868^M
spi3 at pins miso=GPIO19/mosi=GPIO23/clk=GPIO18^M
lmic is at spi3, pin cs=GPIO5^M
Sending data...
Sending data...
Guru Meditation Error of type InstrFetchProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00060730  A0      : 0x800f9ca1  A1      : 0x3ffd1eb0
A2      : 0x3ffbe1a8  A3      : 0x3ffbc1c0  A4      : 0x3ffbea5c  A5      : 0x400f9a84
A6      : 0x3ffb2078  A7      : 0x00000014  A8      : 0x8016cc6c  A9      : 0x3ffd1e90
A10     : 0x3ffbe1a8  A11     : 0x00000001  A12     : 0x3ffd175c  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x00000000:0x3ffd1eb0 0x400f9ca1:0x3ffd1ed0

Rebooting...

```

jolivepetrus, 2017-06-09:
The LMIC stack is full of asserts that call to the abort function (that generates Guru mediation errors). In most cases this asserts are related to SPI or DIO line problems.

We have nodes using the Whitecat N1 board sending data without any problem. Lua RTOS firmwares between boards are identical, the differences are only related to the pin maps.

Please, send us:

* Your sdkconfig file.
* The commit number of your firmware.
* The connections you use between the ESP32CORE and the RFM96 module.

webfrank, 2017-06-10:
Commit: 891aa690fadd43b92fe2636909f6a79cb6e8ca68

Config:
```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"

#
# Bootloader config
#
# CONFIG_LOG_BOOTLOADER_LEVEL_NONE is not set
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=y
# CONFIG_LOG_BOOTLOADER_LEVEL_WARN is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_INFO is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE is not set
CONFIG_LOG_BOOTLOADER_LEVEL=1

#
# Security features
#
# CONFIG_SECURE_BOOT_ENABLED is not set
# CONFIG_FLASH_ENCRYPTION_ENABLED is not set

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/tty.SLAB_USBtoUART"
# CONFIG_ESPTOOLPY_BAUD_115200B is not set
# CONFIG_ESPTOOLPY_BAUD_230400B is not set
CONFIG_ESPTOOLPY_BAUD_921600B=y
# CONFIG_ESPTOOLPY_BAUD_2MB is not set
# CONFIG_ESPTOOLPY_BAUD_OTHER is not set
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=921600
CONFIG_ESPTOOLPY_COMPRESSED=y
# CONFIG_FLASHMODE_QIO is not set
# CONFIG_FLASHMODE_QOUT is not set
CONFIG_FLASHMODE_DIO=y
# CONFIG_FLASHMODE_DOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE="dio"
# CONFIG_ESPTOOLPY_FLASHFREQ_80M is not set
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
# CONFIG_ESPTOOLPY_FLASHFREQ_26M is not set
# CONFIG_ESPTOOLPY_FLASHFREQ_20M is not set
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_4MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_16MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE="2MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_BEFORE_RESET=y
# CONFIG_ESPTOOLPY_BEFORE_NORESET is not set
CONFIG_ESPTOOLPY_BEFORE="default_reset"
CONFIG_ESPTOOLPY_AFTER_RESET=y
# CONFIG_ESPTOOLPY_AFTER_NORESET is not set
CONFIG_ESPTOOLPY_AFTER="hard_reset"
# CONFIG_MONITOR_BAUD_9600B is not set
# CONFIG_MONITOR_BAUD_57600B is not set
CONFIG_MONITOR_BAUD_115200B=y
[... 388 lines cut ...]
CONFIG_MBEDTLS_HAVE_TIME=y
# CONFIG_MBEDTLS_HAVE_TIME_DATE is not set

#
# OpenSSL
#
# CONFIG_OPENSSL_DEBUG is not set
# CONFIG_OPENSSL_ASSERT_DO_NOTHING is not set
CONFIG_OPENSSL_ASSERT_EXIT=y

#
# SPI Flash driver
#
# CONFIG_SPI_FLASH_ENABLE_COUNTERS is not set
CONFIG_SPI_FLASH_ROM_DRIVER_PATCH=y

```

ESP32    --    RFM96
19                   MISO
23                  MOSI
18                  CLK
5                    NSS
26                 DIO0
25                 DIO1

jolivepetrus, 2017-06-12:
We can't reproduce this. Tested with your sdkconfig, same commit, your sample code, ESP32COREBOARD, and FRM95 (we don't have a RFM96 module for test).

Have you anything connected to GPIO33?
You can try to connect the RFM96  RST pin to ESP32COREBOARD GPIO27.
Backtrace is only "Backtrace: 0x00000000:0x3ffd1eb0 0x400f9ca1:0x3ffd1ed0"?
You can try also to do this wiring for power RFM96:

jolivepetrus, 2017-06-12:
Con you confirm that you are using a RFM96?

RFM96 is a 433/470Mhz RF Transceiver, and LMIC only supports 868 / 915 Mhz.

What is your location?.

webfrank, 2017-06-12:
Sorry RFM95, the 868MHz, I'm in Italy

webfrank, 2017-06-18:
Found the problem, board reboots when I plug MacBook power supply, don't know why but it cause interference.

closed 2017-06-18 by webfrank