whitecatboard/Lua-RTOS-ESP32 #56: Add a section to the doc regarding how to transfer lua files

fjolliton, 2017-07-24:
Hi.

I'm playing with an ESP32 Thing from Sparkfun. I like the idea of using a lightweight scripting language to simplify programming on this board, so I installed Lua RTOS on it.

However, to my surprise, I found nowhere how to transfer a Lua file from a Linux terminal to the board.

I searched every terms I could think of. I read a good part of the wiki (all the pages that seemed relevant). I tried ESPlorer (but that seems targeted only at NodeMCU, not Lua RTOS).

Maybe it's not possible and there is no simple tool for Linux to achieve that. But either ways, I think it might be a good idea to talk about that in the documentation (whether it's feasible or not.)

I don't think the only way is to rebuild the FS image and then flashes it each time. Right?

With NodeMCU, there is a tool (nodemcu-uploader) to download/upload file. I was expecting something similar for Lua RTOS.

How do you transfer a single file (without an IDE)? I can't be the only one wondering about that.

bsuhel, 2017-07-24:
For now the only way i found is to open editor in remote terminal, go to
lua program on windows editor and coppy all program code (with ctrl A) ,
then with right click on linux remote terminal copy the code. Than save
it.... I try to find same solution like ssh terminal......

LD

jolivepetrus, 2017-07-24:
Hi @fjolliton, @bsuhel,

File transfers can be easily using The Whitecat IDE. Whitecat IDE is available at http://ide.whitecatboard.org. This IDE allows the programmer to develop programs for Lua RTOS using Lua, or blocks.

You can find relevant wiki for the IDE at https://github.com/whitecatboard/whitecat-ide/wiki

The IDE needs The Whitecat Create Agent, a small piece of software that runs on your computer, and allows the communication beetween a Lua RTOS device and The Whitecat IDE.

We can reuse some parts of the The Whitecat Create Agent for create a command line tool for transfer files to a Lua RTOS board if it's really required by the community, but using The Whitecat IDE you have more functionalities:

* Get access to the Lua RTOS console through the IDE without using external programs.
* When you plug a board the IDE checks for a new firmware version, and you can upgrade the board from the IDE.
* Lua online editor, if you program using Lua.
* etc ...

Please, give me your feedback about our proposal, and the need to develop or not to develop the command line tool.

Best regards,

Jaume

fjolliton, 2017-07-24:
Just to clarify things, I was looking for a very simple tool. I do not want to download an entire IDE, or figure how to extract the part that manage file transfer.

The IDE is probably nice and convenient for a lot of tasks (and your website seems to indicates that lot of efforts is put into it, which is great for the future of theses boards), but I'm used to Emacs and I use terminal for automating everything. So I just made a tool in Python in the past hour to transfer files myself. I put the source at http://tuxee.net/lua_rtos_transfer.py.txt for anyone interested. It's barely functional, but it works for my specific case (transfering a Lua file from my PC to the board in the most easiest way possible).

A command line tool seems mandatory from my point of view. Not everyone use an IDE.

Thanks for your feedback.

jolivepetrus, 2017-07-25:
This issue is now an enhancement

jolivepetrus, 2017-07-25:
@fjolliton, @bsuhel,

Please take a look at this new repo, with first version of the requested tool.

List files in /examples directory:
./wcc -p /dev/tty.SLAB_USBtoUART -ls /examples

Download system.lua file and store it as s.lua in your computer:
./wcc -p /dev/tty.SLAB_USBtoUART -down system.lua s.lua

Upload s.lua file and store it as system.lua in your board:
./wcc -p /dev/tty.SLAB_USBtoUART -up s.lua system.lua

Note: it's a go program and works on Mac OS, Linux and Windows. We can provide build versions if you send us your private mails until binaries will not be published online (we must test it before).

jolivepetrus, 2017-07-25:
Sorry, my mail:

jolive@whitecatboard.org

fjolliton, 2017-07-25:
Sounds great! I will try it later today.

jolivepetrus, 2017-07-25:
You can upgrade the board with last available firmware from the tool.

./wcc -p /dev/tty.SLAB_USBtoUART -f

fjolliton, 2017-07-25:
That seems to work nicely.

Two little issues:

 - if I omit the `-p` option, the process panics because of a segmentation violation
 - the `-down` command doesn't take into account the target filename, and use the remote (board) filename as the local name (`-down foo.lua bar.bar` retrieve the `foo.lua` file from the board, but store it on disk as `foo.lua` not `bar.lua`, because `src` is reused on line 308 instead of `dst` in `main.go`)

I can open an issue on the  `whitecat-console` project if you want.

Anyway, this is exactly the kind of convenient tool I was expecting. Thanks :)

jolivepetrus, 2017-07-25:
Thanks for report this,

Solved in https://github.com/whitecatboard/whitecat-console/commit/6e425be95c898b8e4c69d5bfe4fbe34008447271

Please, give us an star in Lua RTOS github, if you want, and you think that Lua RTOS is useful. This helps the Lua RTOS to be better known and reach more people. ;)

closed 2017-07-25 by jolivepetrus