whitecatboard/Lua-RTOS-ESP32 #57: Delete files?

bsuhel, 2017-07-28:
We cant delete lua or xml or any files from Whitecat IDE.

Only way is to end wccagent.exe and start putty. From "normal" editor than all command work normaly.

Why hidening some stuf and why just not opened everything.? 

Where to find saved lua or xml files on computer, same problem....

If i use Whitecad editor some command are filtered(not working?) why that?

Thing is totaly useless in school if you block on that way.....

jolivepetrus, 2017-07-28:
Hi @bsuhel,

Thanks for report this.

The Whitecat IDE is in alpha version. The reason to publish this tool to the community were to obtain feedback from the community.

Delete lua or xml or any files from Whitecat IDE: we will include this soon.

Where to find saved lua or xml files on computer: this is  complex, because is browser-dependent and your desktop platform. From browsers we only have access to a sanboxed file system that is managed by the browser.

os.shell(false): The Whitecat IDE disables the shell for now.

BLE: this is an amazing feature, but complex. We want to include BLE support, but probably we will have a call to specs to the community before. Please, don't doubt to send us your specs.

DHT22: we will include this soon. We have commitments with some universities and schools for include support for 37 sensors/actuators by Novembre/2017.

I will close this issue now, and recreate-id in The Whitecat IDE project.

jolivepetrus, 2017-07-28:
New issues:

https://github.com/whitecatboard/whitecat-ide/issues/7
https://github.com/whitecatboard/whitecat-ide/issues/6
https://github.com/whitecatboard/whitecat-ide/issues/5

closed 2017-07-28 by jolivepetrus

bsuhel, 2017-07-28:
> New issues:
>
> whitecatboard/whitecat-ide#7
> whitecatboard/whitecat-ide#6
> whitecatboard/whitecat-ide#5

Thank you for respond.

:).

jolivepetrus, 2017-08-14:
Hi @bsuhel.

DHT22 sensor is supported from commit# https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/b4e0543eda011405567f6be3040f6c3886deebc1.