whitecatboard/Lua-RTOS-ESP32 #60: lsntp_stop -> net_check_connectivity

the0ne, 2017-08-01:
What is the reason for calling net_check_connectivity() inside lsntp_stop?

Example:
1) network connection is good
2) start sntp
3) network connection changes to "bad" (wifi problem, etc.)
4) sntp cannot be stopped

I believe that the call to net_check_connectivity() in function lsntp_stop(lua_State* L) could be removed.
What do you think?

jolivepetrus, 2017-08-03:
Hi @the0ne, sorry the delay in my answer.

I do not remember exactly why I put that check in the code. Have you try to remove the check and reproduce your example?.

I think that this check can be removed.

the0ne, 2017-08-07:
i'll clean it out once i can successfully run a build again.
seems something is wrong with the current source...

closed 2017-08-07 by the0ne