whitecatboard/Lua-RTOS-ESP32 #61: UART2 not working?

tcoram, 2017-08-02:
Hello!
I am reading from a GPS device that is streaming NMEA lines. The following works consistently (from the console):
```
uart.setpins(uart.UART1,4,5)
uart.attach(uart.UART1, 9600, 8, uart.PARNONE, uart.STOP1)
uart.lock(uart.UART1)
uart.read(uart.UART1, "*el", 1000)
```
But, if I reboot and change UART1 to UART2, I can't read data.
```
uart.setpins(uart.UART2,4,5)
uart.attach(uart.UART2, 9600, 8, uart.PARNONE, uart.STOP1)
uart.lock(uart.UART2)
uart.read(uart.UART2, "*el", 1000)

```
No errors and attach shows:
 
```
uart2: at pins rx=GPIO4/tx=GPIO5
uart2: speed 9600 bauds
uart2: interrupts enabled

```
I've even tried single character reads with no success.

jolivepetrus, 2017-08-03:
Hi @tcoram,

Thanks for report this.

This is solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/aa6843aeab7d6a1b461ee0628191385f7920a2fe.

Please, try again and give us your feedback.

tcoram, 2017-08-03:
Works perfectly. Thanks! :)

closed 2017-08-03 by tcoram

jolivepetrus, 2017-08-03:
Perfect!

Please, take in mind that since commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/099e77160a1153d2537574ed6421758fb3808ec0 UART NMEA 0183 compatible GPS sensors are supported through the sensor module. Here is an example:

```lua
gps = sensor.attach("GPS", uart.UART1, 9600, 8, uart.PARNONE, uart.STOP1)
while true do
   lon = gps:read("lon")
   lat = gps:read("lat")
   sats = gps:read("sats")
   print("lon: "..lon..", lat: "..lat..", sats: "..sats)
   tmr.delayms(500)
end
```
Relevant wiki is at https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/GPS-SENSOR since today. Sorry, this feature was not documented. 

Try it if you want.

Please, give us an star in Lua RTOS github, if you want, and you think that Lua RTOS is useful. This helps the Lua RTOS to be better known and reach more people. ;)