whitecatboard/Lua-RTOS-ESP32 #62: Improved TLS support for MQTT

andreas23, 2017-08-07:
I've recently been trying to use the TLS support in MQTT.

First the good news: the stack is able to establish a secure connection to my MQTT broker (mosquitto in this case), and to send messages.

Bad news: I get this error message:

`SSLSocket error WANT_READ/WANT_WRITE in SSL_read - getch for socket 16`

Also, I'd like to suggest support for server certificate validation, as well as client certificates, as a feature request.

the0ne, 2017-08-07:
hi @andreas23 
will be happy to test your pull request.

andreas23, 2017-08-07:
Looked into it. So support for certificate validation was removed here:

https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/bfaca2f6c818fe54673da22c07d6b9c7fc659061#diff-35574f529fe1fc96033be874bd6aa50dL510

Because the openssl wrapper for mbedtls, which is part of esp-idf, doesn't support the functionality. Right?

the0ne, 2017-08-08:
True.
Should be possible to add (non-wrapper) mbedtls calls to add the verification.

the0ne, 2017-09-27:
@andreas23 do you need advice to implement the required changes?

the0ne, 2017-10-03:
implemented in #86

jolivepetrus, 2017-10-16:
@andreas23, @the0ne,

For us this issue can be closed. Our tests about TLS support are satisfactory.

the0ne, 2018-01-24:
@jolivepetrus I think the issue can be closed as TLS and server cert verification works

closed 2018-01-24 by jolivepetrus