whitecatboard/Lua-RTOS-ESP32 #66: SENSOR for PIR

wjcarpenter, 2017-08-13:
Hi:- 

The wiki invites suggestions for other types of sensors to implement. I would be interested in an implementation for infrared motion detectors (PIR). There are only one or two in common use, and by far the most common is a module with a BISS0001 sensor.

Thanks

jolivepetrus, 2017-08-14:
I can't find any BISS0001 module on ebay. Dou you have any BISS0001 module url for buy one?

wjcarpenter, 2017-08-14:
Sorry, I think I caused some confusion. The actual sensor is BISS0001, but probably no end-user buys those. Instead, there is a commonly available component that has the sensor and supporting circuitry. The component is usually called HC-SR501 (first link below) or HC-SR505 (second link). I think they are both using the same sensor. The HC-SR501 has a couple of pots for adjustment and also a jumper. The HC-SR505 is in different physical shape and doesn't have the pots or jumper.

http://www.ebay.com/itm/1-PIR-Senser-Infrared-IR-Switch-Module-Body-Motion-Sensor-for-Raspberry-Pi/252671306655

http://www.ebay.com/itm/HC-SR505-Infrared-PIR-Motion-Sensor-Precise-Infrared-Detector-Module-Arduino-MCU-/322458461986

Here's a nice write-up on how they operate, etc: https://cdn-learn.adafruit.com/downloads/pdf/pir-passive-infrared-proximity-motion-sensor.pdf

jolivepetrus, 2017-08-16:
@wjcarpenter, thanks for the information. We have order some sensors.

the0ne, 2019-07-18:
@jolivepetrus what's the current state of this? 
* ongoing or
* finished so we can close this issue?