whitecatboard/Lua-RTOS-ESP32 #70: info about transferring files

wjcarpenter, 2017-08-15:
Is any info on the way for this empty wiki page? https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/How-to-transfer-files

I know that the combination of the web IDE and the agent can do it, but I'm looking for an easy way to do it from local files. I had a quick look at the agent code to see if there was an easy way to write a local program that talks to the agent or directly to the board the same way the IDE does it. It looked like enough work that I wanted someone else to do it. :-)

So, is there already some easy pipeline for doing it? (Feel free to tell me that there is a way to have the boards filesystem appear as something mountable on my PC. I wouldn't object to that at all :-))

jolivepetrus, 2017-08-15:
Hi, @wjcarpenter,

Please, take a look to the following closed issue:

https://github.com/whitecatboard/Lua-RTOS-ESP32/issues/56

This will be documented in the wiki as soon as possible.

wjcarpenter, 2017-08-15:
Fantastic. Just the sort of thing I was looking for.

closed 2017-08-15 by jolivepetrus