whitecatboard/Lua-RTOS-ESP32 #72: Bit module

gjongenelen, 2017-08-31:
I'm trying to receive messages via 868MHz. The only problem I'm facing is the checksum of the package. 

To verify the checksum, I need to use bitwise operators (BXOR) which can be found in the bit(32) module. As far as I can see this is not included at this moment.

Am I wrong or is there a possibility to compile with bit(32)-module?

jolivepetrus, 2017-08-31:
No, no!!

Lua RTOS includes a Lua 5.3.4 interpreter.  Since Lua 5.3 bitwise are supported directly by Lua without the bit32 module.

Please see:

https://www.lua.org/manual/5.3/manual.html#3.4.2

Lua code is nice Now!! For example you can do:

a = (b &  0xff00) >> 8

gjongenelen, 2017-09-04:
Alright, got it working. It was a bit misleading because my IDE threw errors about bad characters.

Thnx

closed 2017-09-04 by gjongenelen