whitecatboard/Lua-RTOS-ESP32 #74: Q: u8g support

gjongenelen, 2017-09-04:
Is there a u8g module like there is in nodemcu? I'm trying to print text to a SSD1306 screen.

jolivepetrus, 2017-09-04:
Now I have an OLED with SH1106 chipset.
I don't have OLED with SSD1306 chipset.

In Lua RTOS all graphics displays are accessed from a common api (GDISPLAY
module). SH11106 will be included in GDISPLAY soon. Adding new chipset
support it's not dificult.

Best regards,

Jaume Olivé

bhgv, 2017-09-24:
i supported ssd1306 based i2c OLEDs in my fork of Lua-RTOS-ESP32 (SDA=gpio5, SCL=gpio4). you can get it here:

https://github.com/bhgv/Lua-RTOS-ESP32

but it use Adafruit Arduino libraries for i2c, ssd1306 and graphics. as "+" you can use other Arduino libraries and Arduino like code inside Lua-RTOS-ESP32 (but not any)

you can see ssd1306 Oled functions inside the Lua class "ssd1306". not all Adafruit functions were released for now, but i'm working on this. you can see the full list directly from Lua-RTOS-ESP32:
```lua
for k,v in pairs(ssd1306) do print(k) end
```

before calling "ssd1306.print(string | number {, string | number})", now should be called "ssd1306.textColor(1)". now after init text color is black ("ssd1306.textColor(0)")

(a little complex scheme to expand with new C modules is in the Lua-RTOS-ESP32)

bhgv, 2017-09-25:
ok, i added a lot of text, fonts and graphical functions from adafruit library into Lua. object ssd1306 and its members.

[ssd1306-lua documentation](https://github.com/bhgv/Lua-RTOS-ESP32/blob/master/doku/ssd1306-oled.md)

gjongenelen, 2017-10-10:
Could you create a PR for this?

bhgv, 2017-10-10:
@gjongenelen 
yes, but i should understand first how to do this

if you want preciselly u8g, i ported it from nodemcu to my fork of esp8266 branch of this os. you can take it there.

jolivepetrus, 2017-10-10:
Hi @bhgv, hi @gjongenelen,

Thanks for your work in the u8g module, but I think that the right is to extend the gdisplay module and add the u8g low level functions on in.

The gdisplay module / gdisplay driver is designed for add new chipsets in a very agile form, and has the advantage that the programmers deals with displays using the same API.

jolivepetrus, 2017-12-04:
Hi @bhgv, @gjongenelen,

Since commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/ced22aee6b46714309f01a88e489a692d186e337 Lua RTOS has support for SSD1306 chipset integrated in the Lua gdisplay module.

See wiki: https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/GDISPLAY-module

closed 2017-12-04 by jolivepetrus

jolivepetrus, 2018-03-12:
Hi @skitsanos,

This is a symptom that you are using a firmware with a bad I2C pin map. Probably you need to change the pin map in kconfig (make menuconfig):

In "Component config -> Lua RTOS -> Hardware -> I2C pin map" check that the pin map is correct for your board.

In "Component config -> Lua RTOS -> Hardware -> Graphic Display -> I2C displays" check that the I2C port which the OLED display is attached is correct.