whitecatboard/Lua-RTOS-ESP32 #75: Bluetooth support?

max1220, 2017-09-06:
As far as I can see, there is no support for bluetooth, but the ESP32 has bluetooth, and the esp-idf supports it as well, so it should be doable without to much of an effort.

jolivepetrus, 2017-09-07:
Yes, I'm agree with you.

Now I'm involving in many things: add support for external GPIO / PWM chips, finish the IDE, release an expansion board for the WHITECAT N1 board with CAN/SDCARD, etc ...

This is scheluded by 4Q 2017, but not started yet.

May be someone from the community can help?

the0ne, 2017-09-28:
@max1220 bluetooth isn't just one function but offers many different "protocols" and use cases. If the Lua RTOS should offer more than only low-level functionality then that would mean to implement such "use cases" instead.
so possibly it would be a step forward if you would specify your use case(s) first.

max1220, 2017-09-29:
I'm actually "porting" this to the SHA2017 badge(Which in it's default micropython firmware does not support bluetooth).
Most useful for most users would be RFCOMM/SPP, as it's somewhat generic, and can easily be used to control an appliance via, e.g. a phone.
I'd personally like to see HID and audio support, but I guess that's a litte more niche.

the0ne, 2017-10-09:
audio (sink) support should be straight forward as there's a sample included in the esp-idf.
ble (gatts) spp server and client are included as samples as well.

i didn't find an rfcomm spp sample in the current esp-idf despite i thought i had seen one before.
the api at esp-idf/components/bt/bluedroid/stack/rfcomm/port_api.c contains a function RFCOMM_CreateConnection that you could use as a starting point

the good news is that a sample can be found at https://esp32.com/viewtopic.php?f=13&t=1722
but the bad news is that **RFCOMM currently**, according to https://www.esp32.com/viewtopic.php?f=18&t=1928&p=9328&hilit=rfcomm#p9328 **doesn't seem to be implemented properly**.

jolivepetrus, 2017-10-16:
I'm agree with @the0ne. We need to know the use cases to provide support for BT in Lua RTOS.

1soproni, 2018-01-24:
I would like to use PBAP and HFP features. Browse in the phonebook and initiate, answer or cancel calls.

chengyuhui, 2018-04-16:
Is there any possibility of GATT support?

galenzhao, 2019-03-29:
@jolivepetrus 
thanks for what you have done.

I think BLE is a very important function for some can using their phones for config or connect their devices.

the0ne, 2019-07-31:
Hi guys,

to summarize, this functionality has been committed to the mainline already.
Some fixes are available for advertising raw data in https://github.com/whitecatboard/Lua-RTOS-ESP32/pull/289

Advertising the data from the RAW data taken from a ESP32 Demo source code now works:

```
bt.attach(bt.mode.BLE)
advData="0201060303F0AB0F094553505f5350505f534552564552"
bt.advertise.start(160, 160, bt.adv.ADV_IND, bt.ownaddr.Public, bt.peeraddr.Public, "808182838485", bt.chann.All, bt.filter.ConnAllScanAll, advData)
```

However, I feel having non-raw advertising could be much more convenient.

megazhuk, 2020-01-20:
Hi friends ✌
Are there any more info about bt module?
As i can see this thread is only one small piece of info about BT module. I can't find any information in wiki and other threads. May be you can make some examples or just give more information?

i want to use BLE to exchange data between 3..5 esp32 modules(with lua rtos) a control them from smartphone. May be you can show me some directions to search more info ?

Thanx guys!
PS. Lua on RTOS is amazing 👍 Great project!

ziotibia81, 2020-03-31:
Hi!

I'm interested on BT support for SPP in master mode, but also in slave.

Like this:

https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/src/BluetoothSerial.cpp