whitecatboard/Lua-RTOS-ESP32 #76: Lua sockets?

max1220, 2017-09-06:
A library providing functionality like lua-socket for opening TCP/UDP connection would help implementing different protocols more quickly(No need for C module).
If the API is similar enough, it might be possible to write a "translation layer" to run unmodified lua-socket applications on the badge (Or at least lessen the porting difficulty).

jolivepetrus, 2017-09-07:
Yes, I'm agree with you.

Now I'm involving in many things: add support for external GPIO / PWM chips, finish the IDE, release an expansion board for the WHITECAT N1 board with CAN/SDCARD, etc ...

Maybe someone from the community can try to see if it is feasible to build Lua Sockecs in Lua RTOS?.

drauschenbach, 2018-01-08:
FWIW, I started [a port of LuaSocket to GopherLua](https://github.com/BixData/gluasocket). I know that Go is not Lua RTOS. But maybe it gives someone the inspiration to see that it is in fact possible to reimagine LuaSocket on a different underlying platform -- thanks in large part to the simplicity of the Lua module architecture.

lu1210, 2018-02-07:
Is there a timeline for supporting UDP and TCP stacks?

jolivepetrus, 2018-02-12:
Hi @max1220, @drauschenbach, @lu1210, @andreas23, @jcwren,

I hope that a Lua Socket port for Lua RTOS will be available in a few days ... I have to solve some issues and implement some missing functions to port this library.

jolivepetrus, 2018-02-14:
Hi @max1220, @drauschenbach, @lu1210, @andreas23, @jcwren,

I have just committed the luasocket library to the master branch. See commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/c318dc8853fdd12df992b9e51b504f8afe854795.

Please, note that currently are some limitations, due to that is required to do some reverse DNS queries from LWIP, to use some of the socket.dns functions. Please, read the commit message to know more about that. If anyone have some idea to do reverse DNS queries in LWIP, please, share it with us.

Please, share with us any feedback / development using our luasocket port, and remember to give us an star in github if you have not done it yet, and you want to do it.

Enjoy!

drauschenbach, 2018-02-15:
Outstanding! This is a huge win, and largely expands the reach of Lua apps that can now potentially address Lua RTOS based edge devices.

closed 2018-02-19 by jolivepetrus

the0ne, 2019-09-24:
Hi @max1220, @drauschenbach, @lu1210, @andreas23, @jcwren, @jolivepetrus,

did anybody of you manage to successfully use luasocket to query a remote http server?

the0ne, 2019-09-24:
Hi @max1220, @drauschenbach, @lu1210, @andreas23, @jcwren, @jolivepetrus,

I can use socket.tcp successful, no problem here:
```
local socket = require("socket")
local tcp = assert(socket.tcp())
tcp:connect(host, port);
tcp:send(data);
local s, status, partial = tcp:receive()
tcp:close()
```

But the **luasocket http** doesn't work:
```
local io = require("io")
local socket = require("socket")
local http = require("socket.http")
local ltn12 = require("ltn12")

local body, code, headers, status = http.request{ 
    url = "http://w3.impa.br/~diego/software/luasocket/http.html",
    sink = ltn12.sink.file(io.stdout)
}
print('code: ' .. tostring(code))
```
The output here is always `code: timeout`