whitecatboard/Lua-RTOS-ESP32 #79: can GDISPLAY share SPI with other devices?

wjcarpenter, 2017-09-11:
I see from Kconfig that GDISPLAY uses SPI3, and I see in the SPI component that I can assign the pins I want to SPI2 and SPI3.

If I use SPI3 for the GDISPLAY device, can I also use SPI3 for other devices? Will the GDISPLAY component cooperate with the SPI component for selecting and deselecting slaves?

jolivepetrus, 2017-09-12:
Yes, you can.

In KConfig, "Lua RTOS -> Hardware -> Graphic display" category you can set the CS assigned to the display.

wjcarpenter, 2017-09-12:
That's good news. Thanks! I figured it would be in there somewhere.

Any info on my second question about using other devices on the same SPI?

jolivepetrus, 2017-09-12:
Yes you can.

You must set a unique CS for each device. If each device has assigned a unique CS in the SPI bus you can attach more than one device at the same SPI bus.

https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus

CS = SS = chip select.

wjcarpenter, 2017-09-12:
Thanks, I know about multiple devices on SPI with different CS pins. I was just asking if the display using the GDISPLAY library would interfere with my direct use of the SPI library. So they won't both try to select different devices at the same time, for example.

If it is true that they work OK together, what is the behavior? If GDISPLAY has selected the display for an operation, and a separate thread uses the SPI library to select another device ... does one of the calls block, throw an exception, or something else? (I know SPI calls should be short, but the overlap is still possible.) Or should I set up some external lock to prevent such overlaps?

closed 2018-05-28 by jolivepetrus