whitecatboard/Lua-RTOS-ESP32 #81: ENC28J60 support?

gjongenelen, 2017-09-22:
Is ethernet implemented in LuaRTOS? Can see that 'net' has 'wf', but no 'en'.

jolivepetrus, 2017-09-22:
Lua RTOS support for now is with ENC28J60 compatible module. This is done because RMMI needs more pins.

RMMI support will be available in the next weeks.

ENC28J60 was removed temporally due to recent esp-idf changes.

The following commit re-enables ENC28J60 support again:

https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/22aaa899dcf100aced4ec6feb1222c6aa44b12a0

It's mandatory to enable ENC28J60 support in KConfig, under the "Lua RTOS -> Hardware -> SPI Ethernet".

gjongenelen, 2017-09-25:
Alright, succeeded in compiling with eth_spi enabled. But do you know what wiring to use? Or didn't you test it by yourself?
There is no documentation on connecting the esp32 with enc28j60 as far as I can see.

jolivepetrus, 2017-09-25:
Our driver is tested with the Pmod NIC100 module.

The wiki has beed updated with additional information about how to setup an ethernet connection and the proposed connections.

https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Net-module#comments-about-ethernet
https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Net-module#ethernet

gjongenelen, 2017-10-10:
Alright, apparently the ENC28J60 totally different from the ENC424J600, as this one doesn't work. Would it be possible to use a LAN8720? I'm using a ESP32-EVB now and it would be nice to use the onboard ethernet.

jolivepetrus, 2017-10-10:
LAN8720 is not planned yet, but it will be available as soon as possible.

jolivepetrus, 2017-10-10:
I have ordered some LAN8720 modules for test.

gjongenelen, 2017-10-12:
You're awesome! Can't wait

jolivepetrus, 2017-10-27:
LAN8720 modules for test have arrived today.
Now I'm searching for time for do that ...

gjongenelen, 2017-11-02:
Alright, will try today!

jolivepetrus, 2017-11-02:
Proposed connections:

[image]

gjongenelen, 2017-11-02:
I'm still tying to figure out my pin mapping from this example https://github.com/OLIMEX/ESP32-EVB/tree/master/SOFTWARE/ETH_PHY_demo

jolivepetrus, 2017-11-02:
Yes, but it's not enougth to power on the module during the reset, and requires to make hardware mods on the lan module. You need to have all the lan module pins in hi-impedance state in the boot phase.

jolivepetrus, 2017-11-21:
We close this issue, because we have not feedback from you. Please, reopen it if you need more assistance.

For your information LAN8720 RMII support has been successfully tested in Olimex ESP32-GATEWAY and Olimex ESP32-EVB boards, that since yesterday are supported by Lua RTOS.

closed 2017-11-21 by jolivepetrus

jolivepetrus, 2018-01-26:
Hi @gjongenelen,

FYI this will be our next board, a Lora WAN gateway based on ESP32 PICO D4.

https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Whitecat-ESP32-LORA-GATEWAY

We use a LAN8720A, and chip enabled goes directly to the 50 Mhz oscillator's enabled PIN. It's a more elegant solution that the proposed above.

Also we have test it without oscillator using the ESP32 50 Mhz clock reference on GPIO17 that can be useful in some hardware designs.

Please don't doubt to be in contact if you need more assistance.

gjongenelen, 2018-01-27:
That looks fantastic! When do you expect to release it? You know anything about the retail price?