whitecatboard/Lua-RTOS-ESP32 #83: Installation Problem

tom-niddatal, 2017-09-29:
I am trying to install LUA RTOS on my ESP32S dev board.
Whatever I tried, I got stuck on the installation process.
Tried it first with the wcc from a MAC. With this MAC I am able to communicate with the board with the Arduino IDE. on cu.SLAB_USBto UART.

I can see in the serial monitor that the board is enteeing the download mode, but wcc has no progress.
I tried the same on Windows, same problem

Any ideas?

regards Tom

jolivepetrus, 2017-10-10:
Mmmmmm ....

I haven't this board for test. Please before running wcc do the following from the esp-idf older:

python components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/.... T --baud 921600 erase_flash

After flash is erased run wcc again.

jolivepetrus, 2017-10-16:
Because we don't have feed back we close this issue. Please, reopen the issue in case of problems.

closed 2017-10-16 by jolivepetrus

brainstorm, 2017-12-30:
I can reopen this issue with a more concrete example while running on OSX:

```
(esp32) brainstorm:~ romanvg$ esptool.py  --chip esp32 --port /dev/cu.usbserial-FTYKHBJT --baud 921600 erase_flash
esptool.py v2.1
Connecting....
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 4.4s
Hard resetting...
(esp32) brainstorm:~ romanvg$ wcc -p /dev/cu.usbserial-FTYKHBJT -f -ffs
Can't connect to any board at port /dev/cu.usbserial-FTYKHBJT.

Available serial ports on your computer:

/dev/cu.lpss-serial2
/dev/cu.Bluetooth-Incoming-Port
/dev/cu.usbserial-FTYKHBJT

(esp32) brainstorm:~ romanvg$ wcc -ports
Available serial ports on your computer:

/dev/cu.lpss-serial2
/dev/cu.Bluetooth-Incoming-Port
/dev/cu.usbserial-FTYKHBJT
```

All other toolchains (arduino, micropython, plain screen/picocom) behave correctly, only `wcc` does not seem to be able to send commands to the `/dev/cu.usbserial-FTYKHBJT` for some reason :/

My board is a generic ESP32 and it is in download mode when trying to do all this on OSX 10.12.6 Sierra with and original FTDI cable:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1 (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_REO_V2))
waiting for download
```

Reconnecting/rebooting/resetting does not help with wcc, all other programs/programmers/frameworks work :/

jolivepetrus, 2018-01-02:
Hi @brainstorm,

Can you give us much information about the FTDI cable / board that your are using, such as the URL links?

Thanks!

brainstorm, 2018-01-02:
The cable is a "ftdi ttl232rg" and it is working properly with Lolins,
Wemos and all other boards I throw at it, so no drivers issue or similar.

The board is a custom esp32 one, but the board is not the problem either
since I can flash esp-idf code, mongoose-os and even micropython.

jolivepetrus, 2018-01-02:
Maybe we can check this issue together, if you are near us ...
We can plan a meeting in Citilab - Cornellà.

Please contact to jolive@whitecatboard.org.

brainstorm, 2018-01-02:
Would love to but I'm currently based in Melbourne, Australia :/