whitecatboard/Lua-RTOS-ESP32 #89: uart attach freezes chip

gjongenelen, 2017-10-06:
When I use uart.attach, the terminal crashes (screen). In a previous version this wasn't a issue. Could this be because of a change in LuaRTOS?

jolivepetrus, 2017-10-08:
Have you a sample code that causes this issue?

gjongenelen, 2017-10-09:
I'm on software version: 3ad08da4a0db69f8cefc74eff2cb4f9b0fdee392

Entering the following command crashes the terminal connection:
`uart.setpins(uart.UART2, pio.GPIO2, pio.GPIO4)
uart.attach(uart.UART2, 9600, 8, uart.PARNONE, uart.STOP1)`

Using different UARTS or pins doesn't work. As soon as I run uart.attach, it crashes after entering 1 character in terminal (screen /dev/ttyUSB0 115200)

jolivepetrus, 2017-10-10:
Thanks for report this issue. Something has changed in esp-idf related to the UART.

Please, try with this commit. It should resolve this issue:

https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/d7341ad85cb58811604855081f34cdc0f5baf13f

gjongenelen, 2017-10-10:
Thanks, this fixed it. Is the problem with the weird indents related to this? Because this keeps occurring.
[image]

jolivepetrus, 2017-10-10:
You need to change 2 setting in your kconfig

Follow this steps:

* From the command line: make menuconfig
* Change this settings
Component config -> ESP32-specific -> Line ending for UART output: set to LF
Component config -> ESP32-specific -> Line ending for UART input: set to LF
* Exit and save changes.
* make flash

closed 2017-10-10 by gjongenelen