whitecatboard/Lua-RTOS-ESP32 #92: new way to register new lua C-plugin

bhgv, 2017-10-10:
in my fork of your old esp8266 branch of lua-rtos i designed a little other system of registering lua C-plugins.

current system links whole code archive without link-time optimisation. resulting rom become bigger than it should.

the way proposed by me is simpler, use one point registering, only C prerocessor used, no need to change any source/header files, no need to link whole code without optimisation. example:

// user_modules.conf
USE_LIB(OS)
USE_LIB(U8G)
USE_LIB(I2C)
USE_LIB(SPI)
// ..and so on

i think this is easier and readabler than define-enable way and more effective as it do '--whole-archive' linker command unneeded

if you interested in this say me, i can send a patch after i moved it to current esp32 branch

jolivepetrus, 2017-10-10:
In the ESP32 version registration for a new Lua module is easier:

MODULE_REGISTER_MAPPED(NET, net, net_map, luaopen_net);

bhgv, 2017-10-10:
not only. you should set to 1 enable-define for each used lua-Clib as it used inside of MODULE_REGISTER_MAPPED macro. and you should define it in 2 or 3 points inside sources. not only in the C files.

this thing
MODULE_REGISTER_MAPPED(NET, net, net_map, luaopen_net);
is unchanged for me. 
BUT to refuse ommiting unused directly variables made by MODULE_REGISTER_MAPPED(...), rotables building is unclear for gcc,  you should disable linker-optimisation and embed whole code archive as it is (' -Wl,--whole-archive' command in components/lua_rtos/component.mk) and it depends to enable-defines (it used inside macroses)

proposed way is clear for gcc. it doesn't depend to disable link-time-optimisation and embed whole code into the rom. and it doesn't use any enable macroses. 

and yes, it should only only a little change macroses in the modules.h header. it's easy

adminwhitecat, 2017-10-10:
Ok, perfect.
Please, make a pull request.

bhgv, 2017-10-11:
hi again. i sent a pr with proposed lua module register system but from account 'bhgv2'.

the0ne, 2019-04-09:
the corresponding pull request https://github.com/whitecatboard/Lua-RTOS-ESP32/pull/93 has been closed.
so I guess we can close this, too?

the0ne, 2019-04-18:
@bhgv I'm closing this now. if it's necessary re-open this issue, please let me know.

closed 2019-04-18 by the0ne