whitecatboard/Lua-RTOS-ESP32 #96: lua rotables have a bad idea in its code.

bhgv, 2017-10-16:
most of lua C-module interfaces internally released as rotables. but rotables has one bad idea. it is numerical indexes of rotables.
example:

just try to read any numerical index of any lua module released as rotable.
```
os[1]
os[2]
```
in general modules do not use numerical indexes, but i desided to read channels of ADC and write PWM channels as
```
adc[0]
adc[1]

pwm[0] = 8
pwm[1] =25
pwm.LED5 = 54
..and so on
```
and got wrong work of C-module rotable.

i checked this problem in rotable algo. do you intersted in this patch?

jolivepetrus, 2017-10-16:
Of course, any patch received from the community is well received and we will be interested in evaluating it.

jolivepetrus, 2017-10-16:
Please, do a PR for it's evaluation.

Mynogs, 2019-03-12:
A little note from me: In my small CN1 project I use normal tables, not rotables, for the libraries. However, these tables are empty except for one metafunction. I use __index to redirect to variable, constant and c functions. So I saved a lot of memory and it's still fast.

bhgv, 2019-03-12:
@Mynogs 
The main idea of ​​ro-tables is to store/read these tables completely (functions, data, structure, keys) in flash, and not in normal memory. The linker allows you to map ordinary C variables to fixed addresses in the flash-rom. thus, if you have something read-only (for example, text messages), then you can work with them practically without using memory in a very simple way. The main secret of rotables is in the .ld file.

the0ne, 2019-04-18:
@bhgv is this issue still open? if not, please close it. thanks!

bhgv, 2019-04-19:
sorry, i thought it's not me who should close this.

closed 2019-04-19 by bhgv