whitecatboard/Lua-RTOS-ESP32 #97: pio.pin.getval alaways return 0

pocampov, 2017-10-16:
Is this the forum to make this question?
```
Lua RTOS beta 0.1 powered by Lua 5.3.4

/ > pio.pin.setval(0,pio.GPIO26)
/ > pio.pin.setval(1,pio.GPIO26)
/ > print(pio.pin.getval(pio.GPIO26))
0
/ >

```

jolivepetrus, 2017-10-16:
@pocampov,

The internal implementation for the Lua pio module configures the GPIO as INPUT or as OUTPUT.
When Lua RTOS boots each GPIO is configured in its default state.
In fact, if you use setval you must configure the GPIO as OUTPUT using pio.pin.setdir(pio.OUTPUT, pio.GPIO26), and prior to use getval you must configure the GPIO as INOUT using pio.pin.setdir(pio.INPUT, pio.GPIO26).

This is the current behavior. Please tell us if you are expected another behavior and we will try to extend the pio module for met your needs.

pocampov, 2017-10-16:
Thankyou jolivepetrus for your answer. I need know the state of a OUTPUT pin in a certain moment.
Is this possible?

the0ne, 2019-04-12:
@jolivepetrus I have skimmed through the source code several times now - there's no really "easy" solution that I can think of, but maybe I just didn't see it

a simple solution could be to use some lua object as a proxy for the pio:
all setval calls would have to go through that lua object and the object would internally store the pio state. on getval the lua object would just return it's internal pio state...

the0ne, 2019-04-12:
could be something like the following:

```
OutPin = { initialized = false, gpio = nil, value = nil }
function OutPin:new(o)
	o = o or {} -- create table if user does not provide one
	setmetatable(o, self)
	self.__index = self
	return o
end
function OutPin:initialize(gpio, value)
	self.gpio = gpio
	self.value = value
	pio.pin.setdir(pio.OUTPUT, self.gpio)
	pio.pin.setval(self.value, self.gpio)
	self.initialized = true
end
function OutPin:setval(value)
	if self.initialized then
		self.value = value
		pio.pin.setval(self.value, self.gpio)
	end
end
function OutPin:getval()
	if self.initialized then
		return self.value
	end
end

pin = OutPin:new({gpio = pio.GPIO26, value = 1, initialized = true})

print(pin:getval())
pin:setval(0)
print(pin:getval())
pin:setval(1)
print(pin:getval())
```

@pocampov please try if this solves your issue

the0ne, 2019-07-18:
@jolivepetrus would you like to use a lua approach as above or would you like to have this implemented in C? I guess C?