whitecatboard/Lua-RTOS-ESP32 #100: unattach a GPIO

pocampov, 2017-10-25:
Is possible unattach a pin joined a pwd? 

```
pwm27=pio.GPIO27
ledrojo = pwm.attach(pwm27, 10000, 0.2)
ledrojo:start() 

```

i tryed whit
`ledrojo:stop()`
but the GPIO27 still is joined to pwm0

jolivepetrus, 2017-10-27:
@pocampov, thanks for report this,

This feature is available since commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/a1c8c8654de5f253c4a790122bcae21898df43fc.

In your case, for detach the pin:

ledrojo = nil

or

ledrojo:detach()

pocampov, 2017-10-29:
Thanks you for your help.
I tried this in html page:

```

<?lua
pin26=pio.GPIO26
pwm27 = pio.GPIO27
pio.pin.setdir(pio.OUTPUT,pin26)
pio.pin.setpull(pio.NOPULL,pin26)
if ledrojo ~=nil then ledrojo:detach() end
ledrojo = pwm.attach(pwm27, 10000, 0.2)
ledrojo:start()          
?>

```

The first time it works well

But when I reload the page I get this error:
`/prueba.luap:11: gpio27 is used by pwm0
`

jolivepetrus, 2017-10-29:
Have you upgraded to last commit?
In my board, with last commit your example works as expected.

jolivepetrus, 2017-11-21:
We close this issue, because we have not feedback from you. Please, reopen it if you need more assistance.

closed 2017-11-21 by jolivepetrus