whitecatboard/Lua-RTOS-ESP32 #102: Missing licence

guyou, 2017-11-02:
I did not find the licence of the project. Reading some source files, it seems to use EPL.

Perhaps can you add a LICENCE.txt file in order to clarify this topic.

jolivepetrus, 2017-11-02:
The parts developed by The Whitecat team are licensed under an older version of the MIT license, but there are contributions to the project that are released under other licenses. Please, refer to each Lua RTOS part to know how is it licensed.

closed 2017-11-21 by jolivepetrus

jolivepetrus, 2018-01-21:
@guyou,

We have update the Lua RTOS license model to BSD 3-Clause with some additions restrictions. All developments based on Lua RTOS prior to this commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/8d7a5200a3133237c79a9153a2f49066eae941bf are not affected with this new model.

https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/master/LICENSE