whitecatboard/Lua-RTOS-ESP32 #103: Support for HELTEC ESP32 LORA

guyou, 2017-11-03:
Hi,

I wish to experiment the LuaRTOS on a board made by Heltec and hosting LoRa and an Oled display. Can someone help me?

The board is supported by PlatformIO under reference [heltec_wifi_lora_32](https://github.com/platformio/platform-espressif32/blob/develop/boards/heltec_wifi_lora_32.json).

oupula, 2017-11-04:
@guyou Oled display 显示没什么问题，初始化下就行了，LoRa就不懂了，这个型号板子贵，没舍得买

oupula, 2017-11-04:
@guyou u8g2 Module & u8g2 Display Module

jolivepetrus, 2017-11-30:
Hi guys,

The Whitecat team wants tu give support for the most number as boards as possible, but our funds are limited. Can anyone send us a board to make the pin-map and give support for this board?

Thanks!

qbzzt, 2018-01-07:
You can see a pin map at https://github.com/darthm0e/esp32-101/blob/master/Heltec_WIFI-LoRa-32_DiagramPinout.jpg. If I can finish my article about using it this weekend, I'll send you a board when I get paid for it.

jolivepetrus, 2018-01-26:
Hi guys,

FYI this will be our next board, a Lora WAN gateway based on ESP32 PICO D4.

https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Whitecat-ESP32-LORA-GATEWAY

jolivepetrus, 2018-04-09:
Hi guys,

We have no plans to support lora on this board, almost until someone will send us a test unit.
Recently, we have added support for the fipy board, which has a LoRaWAN transceiver, and in which the Lua RTOS lora stack is working.
We close the issue. Please, reopen it in case if someone needs more support.

closed 2018-04-09 by jolivepetrus