whitecatboard/Lua-RTOS-ESP32 #104: run net.wf.scan() crash

oupula, 2017-11-05:
/ > net.wf.scan()

                           SSID  RSSI          AUTH
---------------------------------------------------
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/usr/local/esp/esp-idf/components/heap/./heap_caps.c", line 281, function: heap_caps_free
abort() was called at PC 0x4013a0e7 on core 0

Backtrace: 0x4008c19b:0x3ffe3a50 0x4008c1c7:0x3ffe3a70 0x4013a0e7:0x3ffe3a90 0x40082d16:0x3ffe3ac0 0x400861ad:0x3ffe3ae0 0x4000bec7:0x3ffe3b00 0x4010750e:0x3ffe3b20 0x40113f51:0x3ffe3b60 0x4011fa0d:0x3ffe3b90 0x401141ad:0x3ffe3be0 0x401141cd:0x3ffe3c00 0x401215f5:0x3ffe3c20 0x40113b81:0x3ffe3c40 0x40114356:0x3ffe3cc0 0x40122439:0x3ffe3cf0 0x4011d210:0x3ffe3d20 0x4011d41e:0x3ffe3d40 0x4011d669:0x3ffe3d70 0x40113f51:0x3ffe3da0 0x401141a6:0x3ffe3dd0 0x401141cd:0x3ffe3df0 0x401215f5:0x3ffe3e10 0x40113b81:0x3ffe3e30 0x40114356:0x3ffe3eb0 0x40122439:0x3ffe3ee0 0x4011d6da:0x3ffe3f10 0x400f6443:0x3ffe3f40 0x4010529f:0x3ffe3f60

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:3
load:0x3fff0010,len:4
load:0x3fff0014,len:3336
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:10504
entry 0x40078b54
Booting Lua RTOS...

  /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2017 whitecatboard.org

build 1509828901
commit d062e3615ce4d85307d92006c29a88d9dc6ea611
Running from factory partition
board type GENERIC
cpu ESP32 rev 1 at 160 Mhz
spiffs0 start address at 0x310000, size 512 Kb, partition spiffs
spiffs0 mounted

Lua RTOS beta 0.1 powered by Lua 5.3.4

Executing /system.lua ...
Type: GENERIC
CPU: ESP32 rev 1 rev A1
OS: Lua RTOS, version: beta 0.1, build: 1509828901, commit: d062e3615ce4d85307d92006c29a88d9dc6ea611

/ >

jolivepetrus, 2017-11-09:
Hi @oupula,

As far I can see you are not using the last Lua RTOS version. Please upgrade to last version.

jolivepetrus, 2017-11-21:
Hi @oupula,

We cannot reproduce this issue, and we have not feedback. We close this issue, please reopen it if you require additional assistance.

closed 2017-11-21 by jolivepetrus

qbzzt, 2018-01-07:
I am having the same issue:

```
/ > net.wf.scan()

                           SSID  RSSI          AUTH
---------------------------------------------------
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/root/esp/esp-idf/components/heap/./heap_caps.c", line 274, function: heap_caps_free
abort() was called at PC 0x4013bb53 on core 0

Backtrace: 0x4008c56b:0x3ffcd160 0x4008c597:0x3ffcd180 0x4013bb53:0x3ffcd1a0 0x40082e4a:0x3ffcd1d0 0x40085da5:0x3ffcd1f0 0x4000bec7:0x3ffcd210 0x4010bace:0x3ffcd230 0x4011c4ed:0x3ffcd270 0x4011fa8d:0x3ffcd2a0 0x4011c749:0x3ffcd2f0 0x4011c769:0x3ffcd310 0x40121675:0x3ffcd330 0x4011c11d:0x3ffcd350 0x4011c8f2:0x3ffcd3d0 0x401224b9:0x3ffcd400 0x4011ba5c:0x3ffcd430 0x4011bc6a:0x3ffcd450 0x4011beb5:0x3ffcd480 0x4011c4ed:0x3ffcd4b0 0x4011c742:0x3ffcd4e0 0x4011c769:0x3ffcd500 0x40121675:0x3ffcd520 0x4011c11d:0x3ffcd540 0x4011c8f2:0x3ffcd5c0 0x401224b9:0x3ffcd5f0 0x4011bf26:0x3ffcd620 0x400f637f:0x3ffcd650 0x40105bf7:0x3ffcd670

```

I just burned the firmware today, so it is probably the latest.

```
/ 
> os.version()
Lua RTOS        beta 0.1        1513304905      13640999a8e5cde354b4377cd4d3f1054cefb15f
/ >

```

I am using this board: https://www.amazon.com/gp/product/B076MSLFC9/ref=oh_aui_detailpage_o00_s00?ie=UTF8&th=1 and because I didn't see it on the list I selected Generic (option 17) when asked about the board during firmware installation.