whitecatboard/Lua-RTOS-ESP32 #105: mqtt module not working as described by wiki

aresrepair, 2017-12-12:
[image]

jolivepetrus, 2017-12-12:
Hi @aresrepair,

mqtt.client has 4 arguments as described in the wiki (https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/MQTT-module#client--mqttclientclientid-host-port-secure).

aresrepair, 2017-12-12:
Hi jolivepetrus,
[image]

aresrepair, 2017-12-12:
it works if I get 6 args:
[image]

jolivepetrus, 2017-12-12:
Please, try to update to last Lua RTOS version. Since commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/d89202db5b8760147f9297cc8b5a7c8ea826da83 mqtt. Since this version Lua RTOS has backward compatibility with the API described in the wiki. Arguments from argument 5 will be described soon.

aresrepair, 2017-12-12:
Thank you @jolivepetrus 
good work!
all works fine!

closed 2017-12-12 by aresrepair