whitecatboard/Lua-RTOS-ESP32 #106: net.wf.scan causes a restart with memory error

qbzzt, 2018-01-07:
```
/ > net.wf.scan()

                           SSID  RSSI          AUTH
---------------------------------------------------
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/root/esp/esp-idf/components/heap/./heap_caps.c", line 274, function: heap_caps_free
abort() was called at PC 0x4013bb53 on core 0

Backtrace: 0x4008c56b:0x3ffcd160 0x4008c597:0x3ffcd180 0x4013bb53:0x3ffcd1a0 0x40082e4a:0x3ffcd1d0 0x40085da5:0x3ffcd1f0 0x4000bec7:0x3ffcd210 0x4010bace:0x3ffcd230 0x4011c4ed:0x3ffcd270 0x4011fa8d:0x3ffcd2a0 0x4011c749:0x3ffcd2f0 0x4011c769:0x3ffcd310 0x40121675:0x3ffcd330 0x4011c11d:0x3ffcd350 0x4011c8f2:0x3ffcd3d0 0x401224b9:0x3ffcd400 0x4011ba5c:0x3ffcd430 0x4011bc6a:0x3ffcd450 0x4011beb5:0x3ffcd480 0x4011c4ed:0x3ffcd4b0 0x4011c742:0x3ffcd4e0 0x4011c769:0x3ffcd500 0x40121675:0x3ffcd520 0x4011c11d:0x3ffcd540 0x4011c8f2:0x3ffcd5c0 0x401224b9:0x3ffcd5f0 0x4011bf26:0x3ffcd620 0x400f637f:0x3ffcd650 0x40105bf7:0x3ffcd670

```

I just burned the firmware today, so it is probably the latest.

```
/ 
> os.version()
Lua RTOS        beta 0.1        1513304905      13640999a8e5cde354b4377cd4d3f1054cefb15f
/ >

```

I am using this board: https://www.amazon.com/gp/product/B076MSLFC9/ref=oh_aui_detailpage_o00_s00?ie=UTF8&th=1 and because I didn't see it on the list I selected Generic (option 17) when asked about the board during firmware installation.

jolivepetrus, 2018-01-11:
Please, upgrade to the last commit. Your board's firmware is from 27 days ago. Recent commits solves some issues on wifi.

jolivepetrus, 2018-01-11:
Any possibility to get the schematic for this board?

qbzzt, 2018-01-11:
Thank you. The pinout is at https://cdn.hackaday.io/files/269911154782944/Heltec_WIFI-LoRa-32_DiagramPinout.jpg . I couldn't find a complete schematic.

qbzzt, 2018-01-11:
I don't have the tool chain to compile the firmware, I just use wcc -p -f -ffs. It is still the same firmware it was four days ago, build ...4905. Do you know how often that gets updated?

    Thanks,
          Ori

jolivepetrus, 2018-01-11:
Try again with wcc, somehing was wrong in automatic buids ....

qbzzt, 2018-01-11:
The good news is that wcc's build got updated:

> / > os.version()
> Lua RTOS        beta 0.1        1515661098      707d75430411440a700124de36c44bcc5f8b5ab0
> 

The bad news is the problem still occurs:

> / > net.wf.scan()
> 
>                            SSID  RSSI          AUTH
> ---------------------------------------------------
> assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/root/esp/esp-idf/components/heap/./heap_caps.c", line 274, function: heap_caps_free
> abort() was called at PC 0x4013bc13 on core 0
> 
> Backtrace: 0x4008c517:0x3ffcd1b0 0x4008c543:0x3ffcd1d0 0x4013bc13:0x3ffcd1f0 0x40082e4a:0x3ffcd220 0x40085da5:0x3ffcd240 0x4000bec7:0x3ffcd260 0x4010bb32:0x3ffcd280 0x4011c55d:0x3ffcd2c0 0x4011fafd:0x3ffcd2f0 0x4011c7b9:0x3ffcd340 0x4011c7d9:0x3ffcd360 0x401216e5:0x3ffcd380 0x4011c18d:0x3ffcd3a0 0x4011c962:0x3ffcd420 0x40122529:0x3ffcd450 0x4011bacc:0x3ffcd480 0x4011bcda:0x3ffcd4a0 0x4011bf25:0x3ffcd4d0 0x4011c55d:0x3ffcd500 0x4011c7b2:0x3ffcd530 0x4011c7d9:0x3ffcd550 0x401216e5:0x3ffcd570 0x4011c18d:0x3ffcd590 0x4011c962:0x3ffcd610 0x40122529:0x3ffcd640 0x4011bf96:0x3ffcd670 0x400f63a3:0x3ffcd6a0 0x40105c4f:0x3ffcd6c0

jolivepetrus, 2018-01-11:
I can't reproduce this flashing one of our boards (Whitecat N1 / https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Whitecat-ESP32-N1) using the generic firmware.

May be a hardware problem?

If you scan the wifi networks at your location (with a computer), how many AP's do you see? May be something related to the visible number of networks?

jolivepetrus, 2018-01-12:
Please, try the following:

wcc -p /dev/.... -erase
wcc -p /dev/.... -f -ffs

qbzzt, 2018-01-12:
Trying to do an erase first didn't work. 

There are eight access points accessible from my house. Tomorrow I hope to write a C program that will scan for access points to see if the problem happens in C too (in which case it is a hardware problem).

qbzzt, 2018-01-12:
Scanning for WiFi access points with C code (https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiScan/WiFiScan.ino) works fine, so I don't think it is a hardware problem:

```
scan start
scan done
4 networks found
1: pomeranch (-44)*
2: DIRECT-8C806BE8 (-56)*
3: 2WIRE811-insec (-57) 
4: Linksys37499 (-94)*

```

jolivepetrus, 2018-02-21:
Can't reproduce this issue in all supported boards.
Can you try with the newest versions?.

qbzzt, 2018-02-23:
It works with the latest version.

> / > net.wf.scan()
> 
>                            SSID  RSSI          AUTH
> ---------------------------------------------------
>                       pomeranch   -41      WPA2_PSK
>                  2WIRE811-insec   -47          OPEN
>                 DIRECT-8C806BE8   -56      WPA2_PSK
>                      ATThnTSIpi   -87      WPA2_PSK

    Thank you,
             Ori

bytebang, 2018-05-01:
I am currently facing the same problem, but I have no idea why this reboot occurs after issuing a net.wf.scan()

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3552
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:12056
entry 0x400784e4
Booting Lua RTOS...

  /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1525194564
commit cd33db0aea9c7774a549e48a7f7f687e5d8dbbad
Running from factory partition
board type GENERIC
cpu ESP32 rev 1 at 240 Mhz
spiffs0 start address at 0x310000, size 512 Kb, partition spiffs
spiffs0 mounted

Lua RTOS beta 0.1 powered by Lua 5.3.4

/ > os.version()
Lua RTOS	beta 0.1	1525194564	cd33db0aea9c7774a549e48a7f7f687e5d8dbbad
/ > net.wf.scan()

                           SSID  RSSI          AUTH  CH1  CH2
-------------------------------------------------------------
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/gue/Desktop/LORA/esp-idf/components/heap/heap_caps.c", line 261, function: heap_caps_free
abort() was called at PC 0x400d4a63 on core 0
0x400d4a63: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x400959a7:0x3ffe26c0 0x40095aff:0x3ffe26e0 0x400d4a63:0x3ffe2700 0x4008285a:0x3ffe2730 0x40082c15:0x3ffe2750 0x4000bec7:0x3ffe2770 0x400edc9e:0x3ffe2790 0x400f4ce5:0x3ffe27d0 0x40101fd9:0x3ffe2800 0x400f4f41:0x3ffe2850 0x400f4f61:0x3ffe2870 0x400fcc5d:0x3ffe2890 0x400f4915:0x3ffe28b0 0x400f50ea:0x3ffe2930 0x400fdac9:0x3ffe2960 0x400fc5bc:0x3ffe2990 0x400fc89a:0x3ffe29b0 0x400fcad6:0x3ffe29e0 0x400f4ce5:0x3ffe2a10 0x400f4f3a:0x3ffe2a40 0x400f4f61:0x3ffe2a60 0x400fcc5d:0x3ffe2a80 0x400f4915:0x3ffe2aa0 0x400f50ea:0x3ffe2b20 0x400fdac9:0x3ffe2b50 0x400fc850:0x3ffe2b80 0x400d5a27:0x3ffe2bb0 0x400d56eb:0x3ffe2bd0
0x400959a7: invoke_abort at /home/gue/Desktop/LORA/esp-idf/components/esp32/panic.c:648

0x40095aff: abort at /home/gue/Desktop/LORA/esp-idf/components/esp32/panic.c:648

0x400d4a63: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x4008285a: heap_caps_free at /home/gue/Desktop/LORA/esp-idf/components/heap/heap_caps.c:123

0x40082c15: _free_r at /home/gue/Desktop/LORA/esp-idf/components/newlib/syscalls.c:42

0x400edc9e: lwifi_scan at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/modules/net/net_wifi.inc:170

0x400f4ce5: luaD_precall at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/ldo.c:434

0x40101fd9: luaV_execute at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/lvm.c:1267

0x400f4f41: luaD_call at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/ldo.c:499

0x400f4f61: luaD_callnoyield at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/ldo.c:509

0x400fcc5d: f_call at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/lapi.c:1222

0x400f4915: luaD_rawrunprotected at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/ldo.c:142 (discriminator 1)

0x400f50ea: luaD_pcall at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/ldo.c:729

0x400fdac9: lua_pcallk at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/lapi.c:1222

0x400fc5bc: docall at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/lua.c:258

0x400fc89a: doREPL at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/lua.c:423

0x400fcad6: luaos_pmain at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/modules/sys/lua_adds.inc:228 (discriminator 1)

0x400f4ce5: luaD_precall at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/ldo.c:434

0x400f4f3a: luaD_call at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/ldo.c:498

0x400f4f61: luaD_callnoyield at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/ldo.c:509

0x400fcc5d: f_call at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/lapi.c:1222

0x400f4915: luaD_rawrunprotected at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/ldo.c:142 (discriminator 1)

0x400f50ea: luaD_pcall at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/ldo.c:729

0x400fdac9: lua_pcallk at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/src/lapi.c:1222

0x400fc850: luaos_main at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/lua/modules/sys/lua_adds.inc:258

0x400d5a27: lua_start at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/sys/main.c:73 (discriminator 1)

0x400d56eb: pthreadTask at /home/gue/Desktop/LORA/Lua-RTOS-ESP32/components/pthread/_pthread.c:553

Rebooting...
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

bytebang, 2018-05-02:
@jolivepetrus Can you please reopen this issue, or should i file a new one ?

jolivepetrus, 2018-05-03:
@bytebang,

Please, try with commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/d463ae6fffe92d7883e6ddb6a91b2ed685cc0f77.

bytebang, 2018-05-04:
The result with commit d463ae6 is that it doesnt crash now, but it also does not list  any WIFI networks.
(Not even the one which is provided by my mobile phone next to the ESP).

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/ > net.wf.scan()

                           SSID  RSSI          AUTH  CH1  CH2
-------------------------------------------------------------

/ >
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

jolivepetrus, 2018-05-04:
@bytebang,

It is really very strange. This situation occurs only when no AP are present, or the board can't detect any AP (hardware issues), but I have to suppose that this is not your case.

Maybe you build environment is really broken. To avoid situations like that I have upgrade the build system to stop the build of Lua RTOS if the tool chain is not the expected (in the past some issues have been resolving upgrading the toolchain and the full esp-idf).

Please, try commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/7dfaf6a51b6d2d2227212bdcc07ee5113bcb97b8 first. Once upgraded, try to build Lua RTOS again, to see if build stops or not.

If not stops, please, consider to start with a fresh install:

rm -f -r esp-idf-folder
rm -f -r lua-rtos-folder
git clone --recursive https://github.com/espressif/esp-idf.git
git clone https://github.com/whitecatboard/Lua-RTOS-ESP32.git

and build Lua RTOS again

bytebang, 2018-05-10:
I have tried the suggested commit - but without success. There are two different types of results: Either it does not show any WLANs, or the system crashes.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Booting Lua RTOS...
  /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1525955395
commit 7dfaf6a51b6d2d2227212bdcc07ee5113bcb97b8
Running from factory partition
board type GENERIC
cpu ESP32 rev 1 at 240 Mhz
spiffs0 start address at 0x310000, size 512 Kb, partition spiffs
spiffs0 mounted

Lua RTOS beta 0.1 powered by Lua 5.3.4

/ > os.version()
Lua RTOS        beta 0.1        1525955395      7dfaf6a51b6d2d2227212bdcc07ee5113bcb97b8
/ > net.wf.scan()
Guru Meditation Error: Core  0 panic'ed (LoadStoreError)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x400041bb  PS      : 0x00060830  A0      : 0x8000425d  A1       0x3ffe633
A2      : 0x00000066  A3      : 0x00000004  A4      : 0x00000005  A5      : 0x000000c6
A6      : 0x4▒0041a4  A7      : 0x00000000  A8      : 0x400907e0  A9      : 0x3ffe6310
A10     : 0x00000066  A11     : 0x00000004  A12     : 0x00000005  A13     : 0x000000c6
A14     : 0x0000cdcd  A15     : 0x00000001  SAR     : 0x0000001a EXCCUSE: x00000003
EXCVADDR: 0x400907e0  LBEG    : 0x401a6c78  LEND    : 0x401a6c9c  LCOUNT  : 0x00000000

Backtrace: 0x400041bb:0x3ffe6330 0x4000425a:0x3ffe6350Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x40095963  PS      : 0x00060033  A0      : 0x80095b4  A      : 0x3ffe6190
A2      : 0x801a3295  A3      : 0x400041fc  A4      : 0x0000009  A5      : 0x3ffe6210
A6      : 0x3ffc46a7  A7      : 0x3ffc46b0  A8      : 0x80095985  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000009  A13     : 0x3ffe63a0
A14     : 0x3ffca4a8  A15     : 0x00800000  SAR     : 0x0000001c  EXCCAUSE: 0x00000000
EXCVADDR: 0x400907e0  LBEG    : 0x000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x40095963:0x3ffe6190 0x40095b1:0x`ffe6`b0 0x4009Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x400cf148  PS      : 0x00070033  A0      : 0x3ffc45f3  A1      : 0x3ffe5fd0
A2      : 0x5c890000  A3      : 0x0000003  A4      : 0x000▒0009  A5      : 0x3ffe6250
A6      : 0x3ffe6270  A7      : 0x3fc4ba4  A8      : 0x80095919  A9      : 0x3ffe5fb0
A10     : 0x00000035  A11     : 0x3ffe6▒30  A12     : 0xc00958d3  A13     : 0x3ffe6230
A14     : 0x3ffe6270  A15     : 0x3ff6000▒  SAR     : 0x00000015  EXCCAUSE: 0x00000000
EXCVADDR: 0x400907e0  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x400cf148:0x3ffe5fd0 0x40095977:0x3ffe5ff0 0x40095ab1:0x3ffe6010 0x40095c89:0x3ffe6090 0x40095eb0:0x3ffe60b0 0x400822f6:0x3ffe60d0 0x40095960:0x3ffe6190 0x40095960:03ffe61b0 0x40095c89:0x3ffe6230 0x40095eb0:0x3ffe62500x400822f6:0x3ffe6270 0x400041b8:0x3ffe6330 0x400041b8:0x3ffe6350 0x401a3295:0x3ffe6380 0x401a462e:0x3ffe63a0 0x401a7ca4:03ffe63d0 0x4016d9b9:0x3ffe6490 0x4016dad5:0x3ffe64c0 0x4016dea8:0x3ffe64f0 0x40177885:0x3ffe6510 0x401779a0:0x3ffe6540 0x40177a2e:0x3ffe6570 0x40178563:0x3ffe65a0 0x4008e590:0x3ffe65c0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3552
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:12056
entry 0x400784e4
Booting Lua RTOS...
  /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1525955395
commit 7dfaf6a51b6d2d2227212bdcc07ee5113bcb97b8
Running from factory partition
board type GENERIC
cpu ESP32 rev 1 at 240 Mhz
spiffs0 start address at 0x310000, size 512 Kb, partition spiffs
spiffs0 mounted

Lua RTOS beta 0.1 powered by Lua 5.3.4

/ > net.wf.scan()

                           SSID  RSSI          AUTH  CH1  CH2
-------------------------------------------------------------

/ > net.wf.scan()

                           SSID  RSSI          AUTH  CH1  CH2
-------------------------------------------------------------

/ > os.version()
Lua RTOS        beta 0.1        1525955395      7dfaf6a51b6d2d2227212bdcc07ee5113bcb97b8
/ > net.wf.scan()

                           SSID  RSSI          AUTH  CH1  CH2
-------------------------------------------------------------

/ > 
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
I am not able to trigger the core panic on demand. Sometime it happens after a `os.version()` sometimes it happens on the first time, and sometimes it does not crash - but it also does not show any WLAN networks (even if there are some in the range of the board).

jolivepetrus, 2018-05-28:
Hi guys,

Anything changes with our last firmware? https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/7b65ec45f72d3ed6bc00faf91299b587046beaae.

I can't reproduce it with:

Whitecat N1 board
ESP32 thing board
ESP32 core board
FIPY
WEMOS
....

bytebang, 2018-05-30:
I have just pulled the latest version, and it seems to work now. I am able to get a list of WIFI networks and I am also able to connect to one of them. Whatever caused this problem, at the moment it seems to be gone.

jolivepetrus, 2018-05-30:
@qbzzt, @bytebang 

Please reopen the issue if necessary.

closed 2018-05-30 by jolivepetrus