whitecatboard/Lua-RTOS-ESP32 #114: project build error

mgrusik, 2018-02-12:
Tell me what could be the problem? Thank you!

C:/msys32/home/Alex/esp/Lua-RTOS-ESP32/components/http/httpsrv.c:1195:23: error: 'CPU_INITIALIZER' undeclared
C:/msys32/home/Alex/esp/Lua-RTOS-ESP32/components/http/httpsrv.c:1199:45: error: 'cpu_set_t' undeclared

jolivepetrus, 2018-02-12:
Hi, @mgrusik,

Please, build with last commit (https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/059b7b5714dce5dc72f6c3c062fee2ac2bbc0072). This should solve the issue.

mgrusik, 2018-02-12:
Hi, @jolivepetrus !
did not help!

and it should be so?
_pthread.c
_pthread.h

jolivepetrus, 2018-02-15:
Can you send us the error? Screen capture, or something like that?

mgrusik, 2018-02-15:
I yesterday updated the project and all the errors were gone, thank you! 
tell me how to load spiffs_image default

jolivepetrus, 2018-02-15:
To flash the default file system:

make flashfs

mgrusik, 2018-02-15:
[image]

jolivepetrus, 2018-02-15:
I'm trying to solve this issue but I'm not using windows from years and years.
Building Lua RTOS and in general any esp-idf program in windows is too slow.
It is mandatory for you to work in Windows?

jolivepetrus, 2018-02-15:
Hi @mgrusik,

Please, try with this commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/e86d50351ffb35cadd6372e29d98e9cb3cd7ef6a. It was some issues in building Lua RTOS on Windows.

In my setup environment works perfect.

make flash -> build and flash Lua RTOS
make flashfs -> build and flash default flash system

gjongenelen, 2018-02-16:
I'm gonna join this topic because it's related to build errors. After the commit of tonight I'm getting build errors:
`Lua-RTOS-ESP32/components/compat/./gai_strerror.c:54:7: error: 'EAI_BADFLAGS' undeclared (first use in this function)`
This is with the latest LuaRTOS and esp-idf versions

mgrusik, 2018-02-16:
hi, i have ~ 13 minutes compiled

jolivepetrus, 2018-02-16:
@gjongenelen,

Please, proceed as follows:

make clean
make upgrade-idf
make

jolivepetrus, 2018-02-19:
@mgrusik,

Have you test last commit?
All is right?

mgrusik, 2018-02-20:
all right, thank you!

closed 2018-02-21 by jolivepetrus