whitecatboard/Lua-RTOS-ESP32 #116: Timer or clock issue

gjongenelen, 2018-02-13:
Hello,

I need a smaller interval than 500us, which is the max currently supported by the hardware timer.

Is this a hardware limitation?

jolivepetrus, 2018-02-13:
Hi @gjongenelen,

Can you be more precise?
From Lua, which Lua module, from C?

gjongenelen, 2018-02-13:
In the file: https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/master/components/lua/modules/sys/tmr.c

`181	if (micros < 500) {
182		return luaL_exception(L, TIMER_ERR_INVALID_PERIOD);
183	}`

What is the reason behind this?

jolivepetrus, 2018-02-15:
Hi @gjongenelen,

Running ESP32 at 240 Mhz each instruction takes 4.166666666 nanoseconds, so in 1 microsecond you can run 240 instructions.

In previous tests, configuring a hardware timer at 1 microsecond starves the CPU and the switching context is not possible. To avoid this, I put a sanity check into the tmr_setup function in drivers/timer.c to avoid to configure timers at periods lower than 5 microseconds. In the Lua module I set this restriction to 500 microseconds, but this value vas arbitrary and probably we can put a lower value.

Please, if you have time, you can play with lower values in the Lua module, and share with us the results to put a more suitable restriction to this value.

jolivepetrus, 2018-02-21:
Hi @gjongenelen,

Any progress on this issue?

gjongenelen, 2018-02-21:
Not really, I tried tuning the minimum down to 200us. But for my use-case the process of each timer was too long to be able to finish before the new timer was fired off. 

I don't think the micros value is a real limit tho. Just not suitable for my use-case.

jolivepetrus, 2018-04-09:
Requested info where not provided.
Please, reopen the issue if anyone can give additions information.

closed 2018-04-09 by jolivepetrus